motioneye-project/motioneyeos #2451: Direct firmware load for zd1211/zd1211b_ub failed with error -2

3dK, 2020-07-01:
Hi,
Was planning to use MotionEyeOS on RPi with USB WiFi dongle, however for some reason zd2111rw driver is failing to be loaded.
WiFi dongle is detected and wlan0 interface is registered, but not usable.

Is there a different module/driver that I can try/use? Below are the log files.
Thanks

### motionEyeOS Version
20200606 (0.42.1)

### Board Model
Raspberry Pi 1 Model B

### Network Connection
Ethernet
WiFi: zd1211rw

### Log Files
**[root@meye-a28019fc ~]# wpa_supplicant -iwlan0 -c/data/etc/wpa_supplicant.conf**
> Successfully initialized wpa_supplicant
> Could not set interface wlan0 flags (UP): No such file or directory
> nl80211: Could not set interface 'wlan0' UP
> nl80211: deinit ifname=wlan0 disabled_11b_rates=0
> wlan0: Failed to initialize driver interface

**[root@meye-a28019fc ~]# ifconfig -a**
> ...
> wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
>         ether 00:1c:df:df:b2:f0  txqueuelen 1000  (Ethernet)

**[root@meye-a28019fc ~]# lsmod**
> Module                  Size  Used by    Tainted: G
> ipv6                  446464 18 [permanent]
> arc4                   16384  2
> zd1211rw               73728  0
> mac80211              712704  1 zd1211rw
> evdev                  24576  0
> spidev                 20480  0
> sha256_generic         20480  0
> snd_soc_bcm2835_i2s    20480  0
> regmap_mmio            16384  1 snd_soc_bcm2835_i2s
> cfg80211              638976  2 zd1211rw,mac80211
> snd_soc_core          192512  1 snd_soc_bcm2835_i2s
> rfkill                 28672  1 cfg80211
> ...

**[root@meye-a28019fc ~]# dmesg**
> ...
>  usb 1-1.3: Direct firmware load for zd1211/zd1211b_ub failed with error -2
> [ 1463.693288] usb 1-1.3: Could not load firmware file zd1211/zd1211b_ub. Error number -2
> [ 1463.693306] zd1211rw 1-1.3:1.0: couldn't load firmware. Error number -2

starbasessd, 2020-07-01:
Can you shift back to version 20190911 and see if it works there? My
zd1211rw doesn't work in current Raspbian or RaspberryPiOS, and I have
retired them

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

3dK, 2020-07-01:
Same issue:
[   22.056155] usb 1-1.3: Direct firmware load for zd1211/zd1211b_ub failed with error -2
[   22.056192] usb 1-1.3: Could not load firmware file zd1211/zd1211b_ub. Error number -2
[   22.056209] zd1211rw 1-1.3:1.0: couldn't load firmware. Error number -2

starbasessd, 2020-07-01:
How far back in Raspbian/RaspberryPiOS do you have to go to get it to work>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

3dK, 2020-07-01:
I did it long (very long) time ago and at that time I recall installing with apt-get some Realtek drivers to make it work. Not sure which one, though.

If I have to go that way, I would prefer to run a Raspbian OS with MotionEye front-end. The goal was really to have a dedicated machine just for motion capture, hence the MotionEyeOS route.

starbasessd, 2020-07-01:
Although the motionEyeOS does have a lighter footprint on hardware,
unfortunately you may have to go the Raspbian/motionEye route, because it
is so locked down, and missing many utilities needed to add little pieces.
Be sure to use the Raspbian LITE version to minimize the footprint...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

3dK, 2020-07-01:
Was really hopping that a different driver/module could be provided, like in this case https://github.com/ccrisan/motioneyeos/issues/99#issuecomment-330970236.

Meanwhile will have to go with headless Raspbian/motionEye.

3dK, 2020-07-03:
Hi @ccrisan ,
Could you please confirm the package version in latest motionEyeOS for zd1211rw wireless driver?

I burned latest Raspbian OS and initially the usb dongle wasn't working, but after installing latest [firmware-zd1211 (1:1.5-6)](https://packages.debian.org/buster/firmware-zd1211) package (via aptitude), it worked.
Will you be planning to update or provide this driver for Raspberry Pi 1 in future releases?

ccrisan, 2020-07-03:
The firmware will hopefully be updated as soon as we merge in the current BuildRoot version.

starbasessd, 2020-10-23:
@3dK Status?

3dK, 2020-10-26:
> @3dK Status?

Awaiting for firmware merge of [firmware-zd1211 (1:1.5-6)](https://packages.debian.org/buster/firmware-zd1211) package, as explained by @ccrisan in previous reply.

starbasessd, 2020-10-26:
Please try the dev20201026 release and see if it is working.
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-26:
I don't know if this will help you, but it's working for me...
I installed the firmware-zd1211 on a RaspberryPiOS
I compressed the folder created in /lib/firmware/ to the attached file.
I then followed the instructions from #99
I ssh'd into the PI
I then ran
mount -o remount,rw / [enter]
scp name@[ipaddress]:/path to zd1211.tar.gz /lib/firmware [enter]
cd /lib/firmware/ [enter]
tar -xvf ./zd1211.tar.gz [enter]
reboot [enter]
Then stick in the USB WiFi dongle
it should show in 
dmesg [enter]
Set up normally.

[zd1211.tar.gz](https://github.com/ccrisan/motioneyeos/files/5440330/zd1211.tar.gz)

3dK, 2020-11-05:
> I don't know if this will help you, but it's working for me...
> I installed the firmware-zd1211 on a RaspberryPiOS
> I compressed the folder created in /lib/firmware/ to the attached file.
> I then [...]

Thanks @starbasessd, it did worked by extracting [zd1211.tar.gz](https://github.com/ccrisan/motioneyeos/files/5440330/zd1211.tar.gz) drivers into /lib/firmware/ directory!
Will try with latest dev20201026 release and will let you know.

Appreciate everyone's help! Time to revive an old RPi

closed 2020-11-05 by starbasessd