motioneye-project/motioneyeos #2453: Connection issues with two camera's

lodgemoorlad, 2020-07-02:
Hi
I have just set up another ME camera and tried to connect my original one to it. I'm looking for a dropdown after hitting the settings tab to do this, but its not there. It was when I first fitted my SD card and switched on to set up this camera, but now the option to add a camera has gone. 
Any body know why?
Thanks

starbasessd, 2020-07-02:
Need more info:
Make & model of board used
Version of MotionEye/MotionEyeOS
Make and model of Camera
Did you log in as admin or user? Be sure by clicking the little profile on
the upper left, and log in as 'admin' with no password.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lodgemoorlad, 2020-07-02:
Hi Starbassed

Thanks for getting back to me.
Camera it's a pi noir camera Rev 1.3
Latest image version 20200606
The pi is a 3b version 2. ( 2015)
Yes logged on as admin on my new camera.
My original which has been running for at least two years, well like a
trollop, I've now forgotten the PW so that will be coming back to the bench
for an image reburn

But, I was hoping the ME app would still see it.
Thanks

starbasessd, 2020-07-02:
I always try new cameras in Raspbian/RaspberryPiOS just to confirm they are
functional. A few people have found their cameras weren't working after a
lot of hair-pulling.
If you dig in your browser, it probably saved your password there.
Let me know after you get re-set up again.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lodgemoorlad, 2020-07-02:
Really appreciate it
Thanks a lot for trying.
Browser, to make matters worse, my pc bust the other week and I've cobbled
another together, so the browser is a good idea, but sadly  not in this
case. But it's in my note pad for future reference, thanks for that and
yes, soon as I've reimaged the original, I'll be back, as Arnie said.
Cheers

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd