motioneye-project/motioneyeos #2454: max resolution Raspberry Cam HQ

DrKeks, 2020-07-04:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606.

### Board Model

I am using the following board/model: Raspberry PI 3A+).

### Camera

I am using the following type of camera: MMAL

My camera model is: Raspberry Cam HQ

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

I am looking for a setting to get the maximum resolution of the camera image possible. 1fps is ok for me, but I would like to have really sharp images for a timelapse video. The maximum resolution I get is 2000x1504 Pixel, but the camera supports up to 12 MPixel.

starbasessd, 2020-07-04:
It may be a factor of motion/motionEye.
May I suggest you install the latest RaspberryPiOS, then install motion and
motionEye following the instructions here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
IIRC, the latest release of motion/motionEye are still older than the
release date of the HD Camera, and the resolution my not be fully supported
yet.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd