motioneye-project/motioneyeos #2455: 1GB SD card

edwardxml, 2020-07-05:
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

Latest version 20200606

### Board Model

I am using the following board/model: Pi Zero W 1.1

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: detects as MMAL

### Network Connection

My motionEyeOS unit is connected to the network via WiFI

Yesterday I set up a fourth Pi Zero W motioneye camera. This one was different from the others in that it used an old, but previously unused 1GB MicroSD card. The others were all 4GB/32GB. I set it to upload movies based on motion to Google Drive. That works fine for a while then it stops. The camera still works fine but ssh shows the data volume is full. Is 1GB too small to properly rotate space and logs? Will the larger SD cards hit the same issue eventually, or will they recycle space?

I have struggled to find reference to a minimum SD card size.

Many thanks.

edwardxml, 2020-07-05:
PS Logging in through ssh to delete data files and rebooting gets things back together briefly.

edwardxml, 2020-07-05:
Motioneye.log:

ERROR: could not save tasks to file "/data/etc/tasks.pickle": [Errno 28] No space left on device

starbasessd, 2020-07-05:
Yes, space is an issue. I have installed on a 2GB midroSD, but ran into the
same issue. It would appear that to have any 'elbow room', a 4GB is
minimum. Even if you offload all other storage, just logs and such will eat
it up.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

edwardxml, 2020-07-05:
Thanks, I will get hold of another, larger capacity SD card and swap it out then.

closed 2020-07-05 by edwardxml