motioneye-project/motioneyeos #2457: Changed static ip from 192.168 to 172.16 and now the website where i see the raspberry pi won't load

Void-Overflow, 2020-07-07:
Hey there, I've got a raspberry pi zero with motion eye os on it, and I recently changed the static ip from 192.168.1.63 to 172.16.69.63.
 Now when I try to open the web server it won't load. I checked my router for the ip and it is showing it as 192.168.1.101, but I didn't even try yo make my ip that. I was wondering if there is anyway I can fix this. I changed the ip because I was trying to forward the port with an AT&T router, and i thought that maybe if i changed it from 192 and made it something else I could make it remote. I also have had this other issue where my camera didn't detect much motion while it was working.

starbasessd, 2020-07-07:
You could try at the console, but the amount of time and effort required,
it's easier to re-image.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-07-07:
You could also try adding a route on your router that says 172.16. exists
on your local network, but, again, it's faster to just re-image.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Void-Overflow, 2020-07-07:
hello i opened the console but it just says panic rebooting in 310 seconds

Void-Overflow, 2020-07-07:
if i re-image does it save all my data?

starbasessd, 2020-07-07:
No. If you have a linux computer, you can reat the SDCard on it to recover
your data, otherwise, you will be starting fresh.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Void-Overflow, 2020-07-07:
how do i add a route to an att router?

starbasessd, 2020-07-07:
That means networking is totally messed up.
One other option.
Put the SDCard back into the computer you imaged it in, and copy the 4
static networking files back into /boot and see if that gets it back online.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Void-Overflow, 2020-07-07:
what are the 4 static network files??

starbasessd, 2020-07-07:
Contact AT&T Tech Support. I've given up supporting every router under the
sun, sorry.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Void-Overflow, 2020-07-07:
ok thanks for the suggestion

starbasessd, 2020-07-07:
Here's the setup files for Static Access Point
You only need the static_ip.conf file.
edit it using notepad++ (NOT notepad or wordpad)
Change to your normal network and IP address, not the 172.16 number
copy it to the /boot partition.

[ConfigFiles.zip](https://github.com/ccrisan/motioneyeos/files/4887030/ConfigFiles.zip)

Void-Overflow, 2020-07-07:
where is the /boot partition

starbasessd, 2020-07-07:
Are you using Windows, Linux, or Mac to create your SD Cards?
In Windows, it usually shows as D: or E:
It will have files like cmdline.txt, start.elf, fixup.dat, and a folder
like overlays.
In Linux and Mac is usually shows as mmcblk0p1
with the same files.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd