motioneye-project/motioneyeos #2458: problems with Raspberry Pi 4 8 Gig and Motioneyeos

lonewolph, 2020-07-08:
im trying to get motioneyeos working on my Raspberry pi 4 8 gig. When i first booted, I was unable to fully boot because of a error with Start4.elf, but I came across a thread saying to replace Fixup4.dat and Start4.elf with files that were linked and it corrected that issue. However now after getting into the UI I cannot get any of the cameras to work no matter what I do. It was all working previously on a raspberry pi 3
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4893614/motioneye.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4893615/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4893616/boot.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/4893617/motion.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4893618/dmesg.log)

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

Raspberry Pi 4b 8 Gig

### Camera

Logitech Webcam C270
Pi Night vision cam plugged in with ribbon cable
2x  of these https://www.amazon.com/gp/product/B00VFLWOC0/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

### Network Connection
Ethernet

### Peripherals

None

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-07-08:
> dt_b_asin_title_o04_s00?ie=UTF8&psc=1

We are waiting for a new build from @ccrisan to fix pi4-8GB, the new HD
camera, and continuous recording.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-07-09:
> What is the new HD camera?

This one:
https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/
just came out recently.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

darkyellow, 2020-08-14:
Is there any indication for when the fix for the 8gb is going to come out?

starbasessd, 2020-08-14:
Not yet...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

darkyellow, 2020-08-24:
I have tried installing manually, motioneye loads up fine but the standard raspberry pi V2 camera doesn't load. It says unable to load video device after adding. I guess there are more underlying issues that need fixing

starbasessd, 2020-08-24:
Wih RaspberryPiOS installed, do the programs raspistill and raspivid work?
https://projects.raspberrypi.org/en/projects/getting-started-with-picamera/3

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

darkyellow, 2020-08-24:
Yes both commands produce valid output

starbasessd, 2020-08-24:
Are you selecting mmal for the camera in motionEye?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

darkyellow, 2020-08-24:
I wasn't but selecting this works! I had 2 other options (some repeated strangely) to choose from so just choose the first. Thanks for the help

Werks-cpu, 2020-08-24:
> You could instal RaspberryPiOS, and follow the instructions for installing
> motionEye from here:
> https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

I followed the instructions here: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

Everything seed to install fine, but when I “systemctl restart motioneye” I expected something to happen.  I had no errors, 
And there appeared to be no menu entry for motioneyes or anything in the task list. I guess I was expecting camera video
on the desktop. Thought?

darkyellow, 2020-08-24:
You need to use a URL to load the webapp

Accessing The Frontend
After having successfully followed the installation instructions, the motionEye server should be running on your system and listening on port 8765. Fire up your favorite web browser and visit the following URL (replacing [your_ip] with... well, your system's IP address):

http://[your_ip]:8765/
Use admin with empty password when prompted for credentials. For further details on how to configure motionEye, see Configuration

starbasessd, 2020-08-24:
“systemctl status motioneye” shows what?
open web browser and go to http://localhost:8765 and you should see the
WebGUI/Administration page

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Werks-cpu, 2020-08-24:
Status appears good. Connected to the web interface on 8765 with admin and blank pswd. Says no camera configured. I have a Rasp camera v2.1 connected. How do I tell it about that camera? Note: thanks for all the help so far . .

starbasessd, 2020-08-24:
Add a camera, and use MMAL for a camera connected via ribbon cable

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Werks-cpu, 2020-08-24:
I entered http://my_ip:8765/cams/  says not found at the bottom field.

Werks-cpu, 2020-08-24:
First is it a “Simple MJPEG Camera” or “Remote motioneye camera” in top field

Next is the URL http://my_ip/MMAL/

 

This does not seem to work

darkyellow, 2020-08-24:
Go to the main admin page
Select add camera from the combo box drop down at the top left
Select camera type = Local V4L2 Camera
Select Camera = mmal
Select ok button

starbasessd, 2020-08-24:
Stop.
Stand back.
If you have the admin page open, click the 3 lines in the upper left.
Click the empty box just to the right.
Click Add Camera
What kind of camera are you trying to add?
CSI (ribbon cable)? Add an MMAL
USB? Add a V4L2
IP Camera? Depends on the camera.
Let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Werks-cpu, 2020-08-24:
Can’t seem to find my camera

 

 

 

 

Can’t seem to find my camera

Werks-cpu, 2020-08-24:
The line /menu in the upper left don’t have a add camer option, but the line at the top of the page says add camera Click her.

I get :

 

 

My camera is ribbon connected. Camera V2.1

starbasessd, 2020-08-24:
There are:
MMAL  (ribbon Camera)
V4L2    (USB Camera)
Network Camera (http:// and rtsp://)
Remote MotionEye Camera (what it says)
Simple MJPEG Camera
You won't see a specific brand, and you'll have to put the address in if
it's the bottom 3.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-08-24:
That camera is an MMAL camera

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Werks-cpu, 2020-08-24:
Sorry but there are only 2 options for Camera Type:  

              Remote motionEye Camera

              Simple MPEG Camera

Werks-cpu, 2020-08-24:
Im on version 0.42.1 on motionEye

Werks-cpu, 2020-08-24:
Do i have a an old version?  I dont have the option for MMAL

Pete Jacobs
TechnoWerks
 Sent from iPhone X

starbasessd, 2020-08-24:
Creating a test machine, will report shortly.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-08-24:
Did you turn on the camera in raspi-config?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-08-24:
Created new PI with fresh install of RaspberryPiOS
Installed motionEye following these instructions:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
connected to http://localhost:8765 from the Pi Desktop
connected to http://<ipaddress>:8765 from another PC
Clicked on 'Click here to add one', got these options:

[image: image.png]
Sounds like you might have an issue with your install.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-08-24:
BTW, I logged inas 'admin, blank password.
Selected the 3 lines in the top left and got the Setting menu
Under General (even with no camera installed) I got
motionEye Version 0.42.1
Motion Version 4.2.2
OS Version Raspbian 10

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Werks-cpu, 2020-08-24:
The first thing I notice is that on my system:

 

motionEye Version 0.42.1

Motion Version        None

OS Raspbian              10

 

Under Add Camera

 

Remote motionEye Camera

Simple MJEP Camera

 

On Mon, Aug 24, 2020 at 1:00 PM Kevin Shumaker <kevin.shumaker@gmail.com <mailto:kevin.shumaker@gmail.com> >

starbasessd, 2020-08-24:
You have a bad install.
If you check, you missed at least step 2

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Werks-cpu, 2020-08-24:
I went back to the Install page and started over.  It’s a bit jumpy, but it is streaming my camera.

Ooops bumped up the framerate and its working great.

 

Thanks for hanging in and helping.

starbasessd, 2020-08-24:
Good deal. Enjoy.

closed 2020-10-23 by starbasessd