motioneye-project/motioneyeos #2459: RPi3B+ motionEyeOS 20200606: wpa_supplicant problems

tstudebaker, 2020-07-10:
### Preliminary Docs
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

I have also checked that all the suggestions in Troubleshooting/Wireless Connection Problems are OK.

### motionEyeOS Version
I am running motionEyeOS version: 20200606

### Board Model
Raspberry PI 3B+

### Camera
My camera model is: RaspCam 2 8MP v2.1
It functions OK via motionEyeOS

### Network Connection
My motionEyeOS unit is connected to the network via: Router and ethernet cable
Trying to connect it using the wlan of the router.

### Peripherals
I am using the following peripherals that I consider relevant to this issue: USB hard drive.
It functions OK via motionEyeOS and saves pictures.

### Log Files
I consider the following log files relevant to this issue:
I can access/download the log files via motionEyeOS/Expert Settings if needed.
I am not able to decode the entries with sufficient knowledge.

I consider the following facts relevant to this issue:
-RPi3B+ connects to the wlan of the router when booting SD-Raspbian.
-verified that the data/etc/wpa_supplicant.conf is read in reboot by changing the SSID.
-verified with hex-editor that the wpa_supplicant.conf does not include CR LF (0x0D 0x0A) but only LF (0x0A)
-have tried all kind of entries in the wpa_supplicant.conf.
-my laptop and smartphone connect to the wlan OK, as does openplotter/RPi3+.
-the motionEyeOS was burned on SD, the file was motioneyeos-raspberrypi3-20200606.img.
wpa_supplicant.conf was added before the first boot.
-setting the motionEyeOS up was done succesfully via ethernet cable connected to my router.
The same router has the wlan.
-I have tried all kind of wpa_supplicant versions...but apparently not the one that works.

Any ideas?
Is there a position in the log files which tells what really went wrong, for example: net not found, authentication, .conf parameter error, timeout...??

The attached wpa_supplicant.conf file includes the statement: scan_ssid=1, deleting that entry does not help. My wlan is not hidden.

wpa_supplicant.conf hexeditor view:

[image]

boot log:

[image]

starbasessd, 2020-07-10:
How did you create your wpa_sullicant.conf file?
If you used notepad.exe, to create/edit it does not create the file
properly. It is not Linux friendly
If you create it in notepad++.exe it should work correctly.
Can you paste your wpa_supplicant.conf file here?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tstudebaker, 2020-07-10:
wpa_supplicant.conf file:

country=FI
update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
scan_ssid=1
ssid="Ssids_Ssid_Ssid"
psk="PSKPSKPSKPSKP"
}

I created it with notepad, and used hex editor to change CRLF to LF, as I understood that it is the way to UNIX LineFeeds.
I  wonder could the underscores of the ssid-name be the problem?

starbasessd, 2020-07-10:
I have found that non-alpha-numeric characters can cause issues.
You can remove scan_ssid=1 also.
Example wpa_supplicant.conf:
----------------
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID"
    psk="password"
key_mgmt=WPA-PSK
}
---------------------
You may want to consider using notepad++.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tstudebaker, 2020-07-10:
I downloaded Notepad++ and created with it my old supplicant file again and compared it to the original, they are exactly the same.

Problem with the underscores is that I cannot change them, the router box is initialized by my operator.
All I got was a paper holding the name (ssid) and the password (psk) of the WiFi.

It has so far functioned flawlessly with all kinds of different devices, even with by Lolin D1 Mini-project.
Everything I have so far needed has been the ssid/password pair.
Therefore it was a big surprise that motionEyeOS has a problem thereabout, although the network-user-interface of the motionEyeOS asks only the ssid and the password. And the hole motionEyeOS seems very versatile and logical.

But big thanks though so far for your effort starbasessd !!

But again I wonder the log-files, there are 5 different files, and seems to me so far that none of them holds a clear statement why the failed-message is generated?

dmesq.log: something goes wrong judging by the time....

[image]

starbasessd, 2020-07-10:
Did you try my copy of wpa_supplicant.conf with your SSID and password?
Can you try 20200203, or 20190911 for your board?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tstudebaker, 2020-07-10:
I made a copy of your wpa_supplicant with NP++, changed country=US to country=FI though and used my ssid/psk-pair.
Moved it with WinSCP to /data/etc and thus replacing the old wpa_supplicant.
Rebooted RPi and same problem persists ...supplicant: failed, and rebooting and disconnecting the ethernet cable from the router leads to boot cycling.

I can try both versions, I think I start with the older one 2019091.
But that happens tomorrow...

starbasessd, 2020-07-10:
Best bet is to copy the file to the /boot partition on the machine where
you image. Recent versions read the files every boot.
Also, try not connecting the Ethernet cable, force it to use the WiFi from
the beginning. It'll also create better error messages.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tstudebaker, 2020-07-11:
Before I continue with the saga of different motionEyeOS-versions I checked my hardware:
-wlan-strength -22dB, surely enough, and not too much
-current peak consumption 1.45A (as the usb-drive is active), PSU is 5V/2.5A
-voltage minimum 5.05 (as the usb-drive is active)
-SD cards are Maxell 16G microSDHC v10 (some have just number 10)
-bougth also one Kingston 16G microSDHC 100Mb/s

I think at this point PSU is strong enough and SD-cards do not have individual differences that affect my problem

Then the test sequence:
-all versions 1.boot as virgins straigth from balenaEtcher and wpa_supplicant.conf copied to SD.
-wpa_supplicant is as starbasessd suggested and not altered (besides the country and ssd-credentials)
-ethernet cable disconnected
-the RPi3B+ goes to a panic state when supplicant fails and the ethernet cable is disconnected and starts rebooting.
At that point I connect the ethernet cable to get control and via USB-keyboard rebooted or poweroffed the machine.
-HDMI-display is connected to RPi3B+.

First try: 20190911
-1.boot supplicant: done
-hostname/80 visible by my IPscanner on my Laptop (connected to the same wlan as the RPi3B+ is trying to connect)
-wlan connection to motionEyeOS ok, did dump all 5 log files, camera shows picture
-motionEyeOS-Network shows my wlan-SSID
-changing admin password, timezone, hostname and selecting all motionEyeOS-Services on and pressing Apply, motionEyeOS comments the need for rebooting, pressing YES, then the wait circle goes roundandround...after a while browser informs that connection is lost.
-now hostname/80 is not visible anymore
-local HDMI-display shows RPi panic, I connect the ethernet cable, in the next reboot supplicant fails, but ethernet gets up
-using USB-keyboard to log in reveals that the admin password and hostname have changed.
-subsequent reboots and poweroffs show supplicant:fail, and wlan does not get up.

20200203
-1.boot supplicant: fail
-ethernet connected
-KB/reboot
-supplicant: done
-ethernet disconnect
-KB/reboot
-supplicabt: done

and Voila! the RPi3B+ is connected to wlan, motionEyeOS setup works, camera works, shutdown works, number 1 SD-card has seen the LIGHT!!

Number 2 SD-card had a little longer birth time as the 1.boot produced supplicat: done, but the IPscanner did not reveal the RPi until several ethernet/panic sequences later.

But now I have 2 working SD-cards (hopefully), let's see what happens when I try to connect RPi Zeros...

I'll be back, if I need help, Big Thanks again for your support  starbasessd!!!
(thank you very many as Leningrad Cowboys say ;)))

tstudebaker, 2020-07-11:
Cheered too early the success of the version 20200203

After poweroff the two SDs behave as previously.
At boot wpa_supplicant mostly fails, but once in a while about 1 time out 20, it manages to make the connection.

Keeping that connection up the motionEyeOS functions perfectly.
Conclusions:
-wpa_supplicant itself is OK, as it sometimes though very rarely connects
-I do not believe the PSU is the issue (according to my measurements)

Changed the RPi3B+ with another of the same model but without success.
Of the versions I have so far tested this 20200203 is closest to be reliable though still far from it when wifi connection is concerned.
Ethernet has all the time functioned without any kind of problems

Am I really the only one who struggles with this kind of problem??

starbasessd, 2020-07-11:
I don't know. I have a Pi3B+ sitting here in front of me.
I imaged using Win32DiskImager and
https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi3-20200203.img.xz
I copied my wpa_supplicant.conf &  ssh.txt files to /boot partition (E: on
my Win10)
Fire it up, and it works consistently. I use a 32GB MicroCenter MicroSD
card, 'cause that's what I have handy.
I have a 3a PS from RPiFoundation, and a 2.5A PS from MicroCenter/Inland,
and both work consistently.
I even have the Pi3B+ in an aluminum case, and my wifi is 2.4Ghz through 2
walls and a door (if you figure line of sight)
I can connect a PiCam (MMAL) to it directly, or a USB generic web cam, or
connect 5 other IP Network cameras, all without issue.
I would try the following commands:

vcgencmd get_throttled

Should report 0x0
vcgencmd measure_temp
Should report under 60`C (even lower is better.
My home WiFi SSID is alphanumeric only, (no spaces or 'special' characters,
and broadcasting SSID is on.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tstudebaker, 2020-07-12:
I do not what happened but as a last gasp:
-Ilas evening put my RaspbianOS-SD in my RPi3
-connected it to my wlan
-surffed here and there
-put the motionEyeOS-SD back and believe or not but the wpa_supplicant was done.

Now RPi3Bplus connects stable through reboots, overnight poweoffs and even poweroff of the wlan-router.

So I carried on and put up my RPi Zero ....-raspberrypi-20200203.img with the wpa_supplicant.conf.
Did not connect the ethernet cable but booted directly to wlan and it works well.

Now I have a problem with the File Storage, I get continuously message "accessing network share failed: cannot mount network share" after using the Test Share.

I try to save the pictures taken by the Zero in the USBdrive connected to the RPi3B+.
I have not found an answer among the issues.

This I know:
-motioneye.conf of both RPi3+ and Zero include:
smb_shares true
smb_mount_root /data/media
-RPi3+ sees the Zero camera after adding the camera to RPi3+

A few pictures i hope to clarify:

View to my usbdrive:

[image]
The blackened area is the IP-address of the RPi3+.

RPi3+ and Zero services both have all the Services ON:

[image]

One example of my different tests to make the USBdrive visible to the Zero:

[image]
Network Server is just blackened out.

What I have so far done:
-have pressed the Apply always before the Test Share.
-have tried different protocoll services.
-have tried different SMB Protocoll Versions
-have tried as Network Server both the hostname of the RPi3+ and its IP-address (xxx.xxx.xxx.xxx).

Any ideas ?

closed 2020-07-18 by tstudebaker