motioneye-project/motioneyeos #2463: motion detection notification

profileman00, 2020-07-12:
Hello, I have installed motioneye as a service in my ubuntu server and i currently have 3 raspberry zero motion cameras and planning to increase to 10. I have an SMS API in my server and wish to to send notifications such that the information (name, time and date) of the camera initiating the alert should be in the sms like in the email notification which contains all that information. wish to know if i can get that information from motioneye to my API
[image]

starbasessd, 2020-07-12:
Most cell providers allow an email to sms gateway, (xxxyyyzzzz@vtext.com
for Verizon) and you can use that without any major changes.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

profileman00, 2020-07-12:
In my location no cell provider offers such a service. that's why i have to handle the editing of the messages from my python API. I wish to assemble information of the particular camera which triggered the alert from motioneye database as it does with email notification. how can i get the data from the motioneye server to use with my API

starbasessd, 2020-07-12:
Did you look at Call a Web Hook, or Run A Command?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

profileman00, 2020-07-12:
Run A Command -> runs a command in my ubuntu server (like a python script) but i wish to know how i can use that command to collect data from the motioneye service and pass it to a python script

starbasessd, 2020-07-12:
I have not played with it at all, and Google-Fu isn't finding what
variables are available. Maybe @CCRISAN or @JASAW can lend a hand?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

profileman00, 2020-07-12:
thank you very much for your help. i asked a friend who told me how to approach the problem. 

i am supposed to pull data from the database directly after every motion detection. pulling the last data in the database which is supposed to be the most recent one causing the alert. hoping the alerts are logged in the database

closed 2020-07-12 by profileman00