motioneye-project/motioneyeos #2468: motionEyeOS-Pi stream to motionEye-Server

TheDuckMann, 2020-07-16:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Raspberry PI Zero W

### Camera

I am using the following type of camera: MMAL Camera

My camera model is: Raspberry Pi Camera Module 5MP (OV5647)

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

Hello,
first of all, thanks for this really nice and convinient OS!
I am new to the whole motionEye-"game", so excuse this question:
I am running motionEyeOS on a RasPI Zero W as written above. This on it's own works fine. 
But I want this to be exandable. Later I want to have multiple PI's (equiped with a camera) to run in different rooms.
I have a server running on which I installed motionEye via docker. This Container also works as intented. My Problem is the connection between the camera (Pi Zero) and the Server. I want to stream to this server and save the stream on this servers internal HDD. I can't figure out how to do that. If I connect them via remote "motionEye Camera" I can only save to the storage of the Pi. But if I connect them in other ways, I can't set the File Storage on my local server HDD. 
Do I really have to install some kind of ftp on this server or am I missing something? 
Basically I simply want to stream the video to the server and let the server do everything else, like changing settings or saving all the streams to its local drive.
Thanks in advance for any hint and/or help!

starbasessd, 2020-07-16:
I don't have experience with Docker.
That said, if your server can see a local drive, you should be able to put
the path in File Storage
The camera should be set up as Simple MJPEG, not Remote Pi Camera. This
will allow you to save locally to the server.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheDuckMann, 2020-07-17:
Hello Starbasessd,

thanks for your reply! If I configure the external camera as a "Simple MJPEG Camera" I am getting no options! :(

[image]

starbasessd, 2020-07-17:
Request:
Log off the screen you are showing here, and log onto the screen with
admin, and check preferences
Look into the upper left for the head profile, click on it, and log in as
admin
Does this change anything?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheDuckMann, 2020-07-17:
The screenshot I send you was the admin page (logged in as admin) from the server, where I configured the external camera (PI) as a "Simple MJPEG Camera". 
On the Camera (Raspberry Pi Zero W with motionEyeOS) I am getting all the options (logged in as admin):
[image]

starbasessd, 2020-07-17:
And if you delete the camera, you get the full menu back?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheDuckMann, 2020-07-17:
Well there is not much because the server itself doesn't have a camera on it.

starbasessd, 2020-07-17:
Do you get more back than with the camera?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-07-17:
Where did you get your docker image?
Are you using something like HomeAssistant?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheDuckMann, 2020-07-17:
This is my docker-compose.yml File:

```
version: '3'

services:
  security:
    image: ccrisan/motioneye:master-amd64
    container_name: motioneye-security
    ports:
      - "8081:8081"
      - "8765:8765"
    environment:
      - TZ=Europe/Berlin
    volumes:
      - etc_motioneye:/etc/motioneye
      - /mnt/data-drive/motioneye-data:/var/lib/motioneye
    restart: always

volumes:
  etc_motioneye:
  var_lib_motioneye:
```

starbasessd, 2020-07-17:
I want to re-create your environment here.
Can you walk me through how you created your docker?
You can email me directly at starbasessd at gmail dot com
I have a spare Ubuntu 20.04 Desktop or Server VM I can throw at it for the
docker host...
Or I can build whatever flavor of Linux you are using.
I am curious as to why I can only reproduce your screenshot of your server
by logging in as 'user'.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheDuckMann, 2020-07-19:
Have you got my E-Mail? Can you reproduce the Problem?

starbasessd, 2020-07-19:
No, I haven't received any emails.
try starbasessd at gmail dot com

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheDuckMann, 2020-07-19:
I did try again a few Minutes ago..did you get an email?

starbasessd, 2020-07-19:
No, nothing. Nothing in junk, nothing in spam. You are sending to
starbasessd@gmail.com ?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheDuckMann, 2020-07-19:
Yes I did....I don't know whats wrong...!

starbasessd, 2020-07-19:
Post it here. Redact any 'private info' like passwords or IP addresses.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheDuckMann, 2020-07-19:
Its basically what I edited in the post above.

starbasessd, 2020-07-19:
I'm sorry, then. I am not able to get the docker install to work on an
Ubuntu 20.04 fully updated virtual machine with bridged networking.
It shows as running, but doesn't connect to any IP address (localhost, or
network) to access.
I don't know enough of docker to be able to assist. I can pop up a Linux
Virtual machine and have the motion/motioneye service up in less than an
hour (faster if I use a cloned, pre-built VM)
Good luck with docker, it's like trying to deal with HomeAssistant
installs. THere are too many other layers to deal with.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheDuckMann, 2020-07-21:
Thank you for trying! Did you use the correct port, something like: 192.168.178.112:8765? There is no redirect from 192.168.178.112:80.

starbasessd, 2020-07-21:
Yessir. Thanks for asking.
I also checked
netstat -apt
to show ports in use. It never opened 80 or 8765.
That is why I was asking how you set up and ran the docker.
I do support here for most other issues, both for motionEye and
motionEyeOS, just wanted to add docker, but it's not working for me.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheDuckMann, 2020-07-21:
Sorry, I didn't mean to be disrespectful. 

In my case I am running all this behind a reverse proxy, but I didn't want to make it more complicated than it is.
If you want, you could modify your compose file to look something like this:

```
version: '3'

services:
  jwilder-proxy:
    image: jwilder/nginx-proxy
    container_name: proxy
    ports:
      - '80:80'
    environment:
      - TZ=Europe/Berlin
      - DEFAULT_HOST: motioneye.yourDomain.lan
    volumes:
      - '/var/run/docker.sock:/tmp/docker.sock'
    restart: always

  security:
    image: ccrisan/motioneye:master-amd64
    container_name: motioneye-security
    ports:
      - '8081:8081'
      - '8765:8765'
    environment:
      - TZ=Europe/Berlin
      - VIRTUAL_HOST: motioneye.yourDomain.lan
      - VIRTUAL_PORT: 8765
    volumes:
      - etc_motioneye:/etc/motioneye
      - /mnt/data-drive/motioneye-data:/var/lib/motioneye
    restart: always

volumes:
  etc_motioneye:
  var_lib_motioneye:

networks:
  default:
    external:
      name: reverseproxy

```

TheoQc, 2020-09-23:
I have a similar setup : some pi with cameras and a server running the docker image of motionEye.

I can add the cameras from the pis in the server with "Remote motionEye Camera" BUT, the options I see on the server are the ones from the pi. So the path in file storage, is relative to the pi, not the server. That is the problem I would like to solve.

Knowing this, I have the impression that when you added your "Remote motionEye Camera" to your docker image, you used the user (instead of admin) credential to add that remote camera, and thus, seeing the user options of that camera, even though you are logged in as admin to the server.

starbasessd, 2020-09-23:
The fix for the PI Hub showing the PICaam settings is to add the PiCams to
the hub as Network or simple MJpeg cameras.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheoQc, 2020-09-23:
Thanks for the info, indeed it work for the settings to be local to the server.

Unfortunately only one of the 2 cameras seems to work. Raspberry camera is working good, the usb camera (on the same pi) does not work as network camera nor mjpeg, butit does work as a remote motioneye camera.

starbasessd, 2020-09-23:
Are you using the 2nd port (8082) for the 2nd camera? (In network or simple
mjpeg mode)?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheoQc, 2020-09-23:
8082 yes, not sure why it didnt work the 2first time I added it, 3rd time has been the charm for me, now all working :)

Thanks again for the quick help.

starbasessd, 2020-09-23:
Good deal, enjoy.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Resolved

closed 2020-10-23 by starbasessd