motioneye-project/motioneyeos #2473: Motioneyeos getting no picture anymore (admin site works) (camera smybol with a gray circle)

engineerlab, 2020-07-19:
Hey all,

I have the problem that I dont getting any picture. Yesterday everthing works fine. I'm using a raspberry pi camera module. 
Searched on github here with the same issue but nothing helps. Can someone help me out here?

`
[root@meye~]# ls /dev/video*
/dev/video0   /dev/video10  /dev/video11  /dev/video12
`
`
[root@meye~]# v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12

mmal service -727555936.-692046 (platform:bcm2835-v4l2):
        /dev/video0
`
`
[root@meye~]# lsmod
Module                  Size  Used by    Tainted: G
ipv6                  446464 22 [permanent]
8192cu                618496  0
spidev                 20480  0
sha256_generic         20480  0
cfg80211              638976  1 8192cu
snd_soc_bcm2835_i2s    20480  0
regmap_mmio            16384  1 snd_soc_bcm2835_i2s
rfkill                 28672  2 cfg80211
snd_soc_core          192512  1 snd_soc_bcm2835_i2s
snd_compress           20480  1 snd_soc_core
snd_bcm2835            24576  0
snd_pcm_dmaengine      16384  1 snd_soc_core
raspberrypi_hwmon      16384  0
hwmon                  16384  1 raspberrypi_hwmon
snd_pcm                98304  4 snd_soc_bcm2835_i2s,snd_soc_core,snd_bcm2835,snd_pcm_dmaengine
snd_timer              32768  1 snd_pcm
i2c_bcm2835            16384  0
snd                    73728  5 snd_soc_core,snd_compress,snd_bcm2835,snd_pcm,snd_timer
spi_bcm2835            20480  0
bcm2835_codec          36864  0
v4l2_mem2mem           24576  1 bcm2835_codec
videobuf2_dma_contig    20480  1 bcm2835_codec
fixed                  16384  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
i2c_dev                16384  0
i2c_bcm2708            16384  0
bcm2835_v4l2           45056  0
bcm2835_mmal_vchiq     32768  2 bcm2835_codec,bcm2835_v4l2
vc_sm_cma              36864  1 bcm2835_mmal_vchiq
v4l2_common            16384  1 bcm2835_v4l2
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2         24576  3 bcm2835_codec,v4l2_mem2mem,bcm2835_v4l2
videobuf2_common       45056  4 bcm2835_codec,v4l2_mem2mem,bcm2835_v4l2,videobuf2_v4l2
videodev              192512  6 bcm2835_codec,v4l2_mem2mem,bcm2835_v4l2,v4l2_common,videobuf2_v4l2,videobuf2_common
media                  36864  3 bcm2835_codec,v4l2_mem2mem,videodev
overlay               106496  3
`

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
motionEye Version | 0.42.1-- | --
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

### Board Model
Raspberry Pi Zero W

### Camera

Raspberry Pi Camera

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

WiFi

### Peripherals

IOBROKER Port 8081 (another Pi)

### Log Files

I consider the following log files relevant to this issue:

motioneye.log
-----------------------------------
2020-07-19 11:54:02: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:54:12: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:54:22: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:54:32: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:54:42: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:54:52: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:55:02: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:55:12: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:55:22: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:55:32: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:55:42: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:55:52: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:56:02: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:56:12: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:56:22: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:56:32: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2020-07-19 11:56:42: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
-------------------------------------------------------------------------------------------------------------------
[boot.log](https://github.com/ccrisan/motioneyeos/files/4943523/boot.log)

braime, 2020-07-19:
Hi @engineerlab,
I experienced similar behaviour since once week, couple of devices showing no camera feed anymore after running for a loooooong time (by the way thanks to the developer and contributors of this great OS):

- iphoneX - no issue
- samsung tv browser - no issue
- samsung s10 - spinner
- win10 desktop - spinner

I had a closer look and a simple update via the GUI fixed all of my issues and the camera shows up on every device again. Perhaps that fixes your issue already?

Cheers :)

engineerlab, 2020-07-19:
Thanks braime for your response. That was the first thing what comes in mind to do an update but sadly nothing fixed after the update :(

motionEye is up to date (current version: 20200606)

Any more ideas?

aetepi, 2020-08-09:
Just got back to my security cam project after a long break and I'm experiencing the same issue.  I'm unable to see previews in Chrome 84.0.4147.105, Firefox 79.0 (64-bit).  I installed Firefox 72.0.2 (64-bit) and the previews work with no configuration changes.  I also tried Firefox beta 80.0b5 (64-bit) and that was unable to load previews.
My motioneye install is not up-to-date: 20190427, but based on the previous comment this may not be relevant.

starbasessd, 2020-08-09:
Try upgrading to 20190911 if you need continuous recording, or 20200203 if
you don't. Both behave with the latest versions of Chrome, Firefox and Edge.
FIrefox 78 & 79 are working fine here, and is Edge 84, and Chrome 84.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

aetepi, 2020-08-09:
Thanks for the comment starbasessd, upgrading fixed it.  I probably should've done that first, but was just browsing the issues and saw this one that was fairly recent.

starbasessd, 2020-08-09:
There have been several issues related to the browser updates/upgrades that
have 'broken' motionEye, but those two versions seem to keep things working.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-10-23 by starbasessd