motioneye-project/motioneyeos #2474: Recordings are being uploaded as 0kb (empty files)

sts-ryan-holton, 2020-07-19:
### motionEyeOS Version

- Motion Eye OS version: 20190911
- Motion Eye version: 0.41
- Motion version: 4.2.2

### Board Model

Raspberry Pi 4 (2gb model)

### Camera

Raspberry Pi camera, loaded on a Pi Zero, streaming over to the Pi 4.

### Network Connection

Wireless on 2.4ghz, approximately a few metres from the router.

### Peripherals

None are connected

### Log Files

I consider the following log files relevant to this issue:

[boot.log](https://github.com/ccrisan/motioneyeos/files/4944290/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4944291/motioneye.log)

### The Problem

I'm using a Raspberry Pi 4gb with Motion Eye OS in order to act as a hub to consume a stream path from another Raspberry Pi (zero).

I'm using an older version of Motion Eye OS since an issue preventing continuous recording to function correctly isn't working and hasn't recently been fixed, see: https://github.com/ccrisan/motioneyeos/issues/2378

The issue I have, is that some video clips aren't actually being fully recorded, and thus are being uploaded to my FTP server as **0kb**. I have my videos set at intervals of 5 minutes, and roughly 15% of my recordings aren't being uploaded reliably despite the file names being set and the network being uninterupted.

How can I fix this?

[image]
[image]

sts-ryan-holton, 2020-07-30:
@ccrisan @jasaw

starbasessd, 2020-10-23:
Status?

sts-ryan-holton, 2020-10-25:
@starbasessd I've since changed my setup for how my media file are uploaded to my FTP server (it's a Synology), I'm not sure why or how this works but it's not what I initially anticipated, here's what works and what doesn't work...

### Doesn't work...

- **Storage device:** the pi.
- **Upload media files:** turned **on**, and upload service set as SFTP, or FTP.

### Works...

- **Storage device:** network share
- **Upload media files:** turned **off.**

And by **doesn't work"" I'm referring to the fact that it does work, but 15% of my recordings are recorded as empty files (not ideal in a situation where you need to review footage), compared with the network share which seems to be more reliable...

starbasessd, 2020-10-25:
You're using the SMB (Windows share) on the Synology if it works that way.
But, it's working...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2022-01-19:
No comments after more than 60 days. If you are still having issues, please open a new issue.

closed 2022-01-19 by starbasessd

Commander-Crash, 2022-05-20:
This is still a issue. GAWD i hate motion.

MarcoMagana, 2022-05-20:
That happens to me when the SD card is almost full!