motioneye-project/motioneyeos #2475: Stream freezes after a while

Pluettmann, 2020-07-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry PI 3B+ with RPi camera and Rapberry PI 4 with 7" touchscreen

### Camera

I am using the following type of camera: RPi camera v2

### Network Connection

My motionEyeOS unit is connected to the network via: wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

raspberry pi 7" touchscreen

### Log Files
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4962606/dmesg.log)
[messages (1).log](https://github.com/ccrisan/motioneyeos/files/4962607/messages.1.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4962608/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4962609/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/4962611/motion.log)

Pluettmann, 2020-07-22:
So my setup is a rpi3 on the front door with motioneyeOS installed and the raspberry camera v2 attached. Inside I use a rpi4 with the raspberry touchscreen. I use chromium to access the motioneye webpage. The camera always is on fullscreen. Serveral times a day the stream freezes and I have to reload the page to continue streaming. In most cases when the stream froze a gray rectangle is present on the bottom half of the camera view. When the page has been reloaded the stream continues.

I thought it might be due to an inactive tab of chromium, because after 15 minutes the screen goes blank when nobody touches it. I turned it off so the screen is active the whole time. But after an hour or so the stream freezes again and I have to reload the page. In the log files I can't see any errors that might explain this issue.
My new attempt to solve this is to install motioneye on the rpi4 as well and access the local motioneye server. I added a remote motioneye camera from the front door and will see if this does the trick.

starbasessd, 2020-07-22:
If you access from Windows, Mac, or Linux, do you get the same issue?
Try pointing to both the Pi3 and the Pi4.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pluettmann, 2020-07-22:
I will try

Pluettmann, 2020-07-23:
So after more than 9 hours my new attempt seems to be working. With the touch screen blanking after 15 Minutes and reactivating it with a touch when needed the stream is working.

Like @starbasessd suggested I tried to access both pi from my Win-PC. This is what I saw this morning:
[image]
[image]

To recap, the rpi3 is outside running motioneyeOS and the rpi4 is inside running motioneye and takes remote control over the outside cam.

sts-ryan-holton, 2020-07-26:
My stream keeps freezing as well with a similar setup with a Pi Zero with the Pi Cam, along with a Pi 4 as a streaming box. The live stream feed freezes after a few minutes, no idea why.

starbasessd, 2020-07-26:
After the camera has been running a bit, but before it freezes, ssh into
it, and run the following commands, and report the results, please:
vcgencmd measure_temp
vcgencmd get_throttled
If you can ssh into the camera after it freezes, run the same commands and
report those results, too, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sts-ryan-holton, 2020-07-26:
@starbasessd For me, I'm able to SSH into the Pi Zero W, and the results are:

```bash
temp=55.1'C
throttled=0x0
```

starbasessd, 2020-07-26:
Ryan;
Temp is OK, and voltage is good. Was the video frozen at the time?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sts-ryan-holton, 2020-07-26:
@starbasessd Not at the time of testing no, but what usually happens is the stream will freeze, and the resolution would only show half of the photo, cutting off the bottom half, I can just refresh to get up-to-date, but it usually at least once in a 10-15 min video. Could be related to https://github.com/ccrisan/motioneyeos/issues/2474 and for reference, the resolution I'm running at is 1024x576

jasaw, 2020-07-27:
@sts-ryan-holton Can you please give more info on your Pi Zero configuration?
- Which protocol? MJPEG or RTSP?
- What is the frame rate?
- What is the bit rate?

The stream will freeze or cut off at the bottom if the wifi network can't keep up with the amount of data you're trying to push over it (like video stream). Try changing to RTSP protocol first (if you're currently running MJPEG protocol, you'll need to remove and re-add the camera on your server). Also try lowering the frame rate and resolution, see if it makes any difference.

sts-ryan-holton, 2020-07-28:
The protocol is MPEG, at around 24fps, any less makes it incredibly unusable. I've got the image quality of the FNC at around 5%.

sts-ryan-holton, 2020-07-28:
@jasaw For clarity, I've double checked my settings of the Pi Zero W being used as a FNC (Fast Network Camera), this is my settings:

- 24fps
- 1000000 bps (default bitrate)
- MJPEG

The video footage is very "choppy" even with the resolution at 1024x576 at video quality at 6% streaming over to the Pi 4. What can I do to improve this? I've temporarily tried settings the FPS to 15 even though I was never getting getting anywhere near the 24fps, hoping it'll be smoother? Also tried changing the bitrate to 3000000 to try and make it smoother...

jasaw, 2020-07-29:
@sts-ryan-holton Try changing the FNC stream protocol from MJPEG to RTSP which is more efficient at compressing video stream. After changing the stream protocol, you'll need to re-add the camera on your server.

sts-ryan-holton, 2020-07-29:
@jasaw How would I change the protocol? Would I need to first remove the camera and re-add it to do this?

jasaw, 2020-07-29:
@sts-ryan-holton To change your Pi Zero W to RTSP mode, do this:
1. Point your web browser to your Pi Zero W IP address or host name. This will bring up the FNC configuration and live stream.
2. Log in as admin and open the configuration panel on the left.
3. Scroll right down to the bottom, you'll be able to see "stream protocol" under "video streaming" section. The default is MJPEG. Change it to RTSP.
4. After changing to RTSP, you'll need to apply it and restart. You may have to reconfigure the resolution, frame rate etc.
5. Point your browser to your MotionEyeOS or MotionEye server. Remove the existing camera.
6. Add a new camera, select "Network Camera" type, and enter `rtsp://<pi_zero_w_ip>/h264` as the URL, replace the `<pi_zero_w_ip>` with your Pi Zero W IP address or host name. Select TCP mode because you're streaming over wifi, which is too lossy for UDP to work properly.
7. If you manage to add the camera successfully, you'll need to configure the resolution, frame rate, etc to match the video stream coming from your FNC.

Resolution 1280 x 1024, 24 fps, 1 Mbps bit rate shouldn't be an issue.

sts-ryan-holton, 2020-07-29:
Cool, I'll have to try this soon to see if it makes any difference.

Currently the Pi Zero W is in FNC and streams to the Pi 4 2gb which is connected via WiFi, Ethernet is going to be a trouble for my setup.

With the current setup the 15fps and a slightly lower resolution that I changed to didn't seem to make much difference to the freezing, the stream still freezes.

The numbers in the bottom left for FPS show something like:

5.0/10.0

And this is at best, I don't know what the two numbers mean.

I'll try changing the mode and report back @jasaw

sts-ryan-holton, 2020-07-29:
@jasaw Unfortunately, `rtsp://<pi_zero_w_ip>/h26` doesn't work. I get a Connection refused error, even tried with the port at the end of the IP.

jasaw, 2020-07-29:
@sts-ryan-holton You missed the "4" at the end of `rtsp://<pi_zero_w_ip>/h26`

sts-ryan-holton, 2020-07-30:
@jasaw Correction, I've missed it out in my comment but not on my actual setup. Doesn't work, I also tried turning tcp off via the motion.conf file, that doesn't work...

jasaw, 2020-07-30:
@sts-ryan-holton Try entering the RTSP stream into VLC player (under network stream in the player).

jasaw, 2020-07-30:
@sts-ryan-holton Just as a test, maybe try changing the RTSP resolution to 1280 x 1024, RTSP frame rate 15 fps, RTSP bit rate 1Mbps, MJPEG resolution 640 x 480, MJPEG frame rate 5 fps, MJPEG bit rate 800kbps.
Then try the VLC player again.

sts-ryan-holton, 2020-07-31:
I'll have to try this later and report back, not convinced it'll work though? What's the thinking behind that? @jasaw

jasaw, 2020-07-31:
@sts-ryan-holton I'm trying to isolate the problem. I suggested the settings above because I know it works.

sts-ryan-holton, 2020-07-31:
@jasaw Tried updating those settings:

> VLC could not connect to "192.168.1.224:554".

Same error as before

jasaw, 2020-08-03:
@sts-ryan-holton hmm... that doesn't make any sense at all. What URL did you enter into VLC? You have to enter `rtsp://192.168.1.224/h264`.
If it still doesn't work, can you please reboot the Pi Zero W?

sts-ryan-holton, 2020-08-03:
@jasaw I did some tinkering, I confirmed that I added the 'rtsp_uses_tcp off' option to 'motion.conf', with this set I can't even access the Pi Zero W over the network via WiFi on another device, for some reason it only works if I'm on the desktop PC which is connected via cable to the router (the Pi Zero of course isn't)

After retrying adding the Pi Zero as a network camera to the Pi 4 I now get a "No Route To Host" error, I can ping the Pi, but only from the PC.

I tried switching back to MJPEG stream and tried adding it back, and got the same error, "No Route To Host"...

I wonder if it's because the TCP option has been set?

That being said, none of what I've tried so far allows the RTSP stream to work...

I'm curious as to why I need to add 'h264' at the end of the URL? What is that referencing and what's the folder location of that? Because it might not exist

jasaw, 2020-08-03:
@sts-ryan-holton I believe it should work better with `rtsp_uses_tcp on`.

>No Route To Host error

This sounds like a networking issue on your network infrastructure. It is not related to the TCP option or RTSP or MJPEG. In order for your Pi4 or PC to find the Pi Zero W, the Pi4 broadcasts ARP packets, asking who owns IP address 192.168.1.224, but didn't get any response back. The fact that your PC can find 192.168.1.224, but not your Pi4 suggests that you have fundamental issue with your network infrastructure.

The `h264` at the end of the RTSP url specifies which RTSP resource to subscribe to. An RTSP server may support multiple video streams, for example: `rtsp://<IP_addr>/highresolution` and `rtsp://<IP_addr>/lowresolution`. In MotionEyeOS, we only have 1 resource at the moment, which is `rtsp://<IP_addr>/h264`.

sts-ryan-holton, 2020-08-03:
I'm not too sure @jasaw everything was working up until I switched to RTSP, and then not even MJPEG works after that now, I do have Pi Hole running, a DNS level adblocker, but doubt that's causing any issues here. I can't install the latest MotionEyeOS version either since continuous recording doesn't work.

jasaw, 2020-08-03:
@sts-ryan-holton Which camera do you have connected to your Pi Zero W?

sts-ryan-holton, 2020-08-04:
I have the official Pi Camera V2 NOIR, and it's connected just fine. If I switched from RTSP to MJPEG the stream would show, but would freeze every few moments and I'd need to reload the browser to get it to show again, which is the purpose of switching to RTSP, to hopefully improve frame rate and resolve freezing frames @jasaw

jasaw, 2020-08-07:
@sts-ryan-holton I can't the problem here. How much memory did you allocate to the GPU? Try setting it to 256MB for RTSP mode.
MJPEG stream freezes every now and then because MJPEG mode essentially sends each frame as a full JPEG image, which requires a lot of wifi bandwidth.

sts-ryan-holton, 2020-08-09:
@jasaw I'll try changing the memory, not sure about the network bandwidth, it's near the router, with the default bitrates.

starbasessd, 2020-10-23:
Status?

closed 2022-01-19 by starbasessd