motioneye-project/motioneyeos #2476: Second camera keeps freezing

cjh7583, 2020-07-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (0.42.1).

### Board Model

I am using the following board/model: (PI 3B+).

### Camera

I am using the following type of camera: (V4L2, Camera).

My camera model is: (usb2 webcam trust).

### Network Connection

My motionEyeOS unit is connected to the network via: (WiFi, ).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. sandisk usb stick)

### Log Files

I consider the following log files relevant to this issue:

Fresh build of motioneye but still my camera keeps locking up. I have 2 usb cameras attached and one of them just locks up at random with the last picture displayed, no grey screen of camera n/a but the frozen last image. The other one continues to work fine.
[Logs.zip](https://github.com/ccrisan/motioneyeos/files/4970965/Logs.zip)

cjh7583, 2020-07-31:
Any suggestions on this one?

starbasessd, 2020-07-31:
Is it always the same camera?
Obviously, check the camera itself. Install only one USB camera and test,
then use the other one (by itself) and test.
Change USB port(s) around. Does it follow the camera, or does it stay at
the same port (freezing)
Do you have anything else connected via USB?
Are they the same model? I have issues of all sorts, when I connect 2 TIGA
Generic DashCams to the same PI or PC. Sometimes they are discovered
correctly, but have issues with video after a while, sometimes they aren't
discovered correctly, or at all. Plug them into separate PIs or PCs and
they work fine.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cjh7583, 2020-08-07:
Hi I have run the camera off a usb hub and has been fine for last week now, hopefully solved the issue.

cjh7583, 2020-08-31:
Mini update - camera still locks up once a week or so but seems its when the sd card gets too full...even though auto delete after 5 days enabled, I still have to occasionally go in and clear out the clips.

closed 2020-08-31 by cjh7583