motioneye-project/motioneyeos #2482: Monitor MotionEye Cams on Television

tech101us, 2020-08-02:
I'm running MotionEyeOS as a VM based on Debian 10. Of all the Linux based NVR solutions out there, I like it the best (with perhaps the one exception that it doesn't seem to support detection of ONVIF cameras).

Here's my "issue"...
Are there any solutions out there to display the MotionEyeOS Cameras on a television? I've got a few Raspberry Pi 3 B+ devices sitting around unused, as well as a Google Chromecast. Wife wants to monitor the cameras in the kitchen, so I'm looking for a solution.

Thanks in advance...

starbasessd, 2020-08-02:
I have a Pi3B+ with the Official Pi 7" screen in my kitchen running
Raspbian and a couple of scripts that toggle between the 5 streaming ports
of my MotionEye server on the display.
It could easily be used on a TV (HDMI out instead of the DSI out)
It auto starts and shifts to rotating the streams on boot. Rotates at 5
seconds per stream (adjustable)
If interested, let me know, I can put together the instructions to put on
here.
Doesn't require motionEye or motionEyeOS on the display machine, just a
Raspbian Desktop. Can run off of a USB Boot stick, too (how mine runs) to
save wear and tear on a SDCard
Let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd