motioneye-project/motioneyeos #2483: Resolution Scale - 1080p

RichBuck89, 2020-08-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING]

I confirm that I have read the [FAQ]

### motionEyeOS Version

I am running motionEyeOS version: 0.42.1

### Board Model
PI4 

### Camera

I am using the following type of camera: ( MMAL, Network Camera, .

My camera model is: (insert your camera model, e.g. Logitech C290).

MakerFocus Raspberry Pi4 Camera IR
https://www.amazon.com/gp/product/B07DPNDP6R/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
### Network Connection

My motionEyeOS unit is connected to the network via: (Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

POE Hat
### Log Files

Hello, 

I don't understand why the image doesn't scale on my Motioneye camera. When I use 1080p half of the image is cut off. When I use 1600 x 1200 I get the full view. 

1080P
[image]

1600 x 1200

[image]

I'd perfer to use 1080p but it's cutting out 1/4 of the overall image. I've played around with the collumns and virtal fit frames. 

Is this a limitation on the camera that I purchased or an issue with MotioneyeOS?

jasaw, 2020-08-03:
I believe this is an issue with Raspberry Pi GPU firmware. I'm not sure what we can do about it. You could try raspivid to see if it behaves the same way. If it does, you could report the problem to the Raspberry Pi github project.

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd