motioneye-project/motioneyeos #2485: Cant set admin/surveillance password

grrcro, 2020-08-04:
### motionEyeOS Version
I am running motionEyeOS version: 20200606.

### Board Model
I am using the following board/model: Raspberry Pi Zero W.

### Camera
I am using the following type of camera: MMAL.
My camera model is: (Raspberry Pi Camera V2.

### Network Connection
My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
n/a

### Log Files

I consider the following log files relevant to this issue:
[boot.log](https://github.com/ccrisan/motioneyeos/files/5022359/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5022360/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5022362/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5022363/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5022364/motioneye.log)

I have successfully changed a few of the settings, static IP, hostname, surveillance username etc, but cant seem to make the admin or surveillance passwords stick after a reboot. Is there something i'm missing or is this a bug?

starbasessd, 2020-10-23:
Did you resolve this?

grrcro, 2020-10-23:
Hi, nope, I tried a few things then left it. 

In the end I used RPi OS Lite, wired up a PIR sensor module to the GPIO and made a few bash scripts to get the capture trigger and auto upload functionality. Rclone handles the Dropbox upload no bother.

closed 2020-10-23 by starbasessd