motioneye-project/motioneyeos #2486: Upnp representation URL

JoanMCD, 2020-08-06:
### Preliminary Docs
Feature request of OS implementation.

Hi, it's possible to add a Upnp presentation function to MEOS to find the Cameras (there Website) more easily in the (Windows) Network Connections Panel?

This also give the possibility to run the Cameras with dynamic IP and still find them!

Here I request only the Upnp representation URL/mechanism - not an implementation as an Media Device !

Some pictures here:
https://github.com/xoseperez/espurna/issues/423

Ciao, Joe

starbasessd, 2020-08-06:
> xoseperez/espurna#423 <https://github.com/xoseperez/espurna/issues/423>

Why it's not a Good Idea(tm):
https://www.varonis.com/blog/what-is-upnp/
UPnP is a security hole that can open your router to many attack vectors.
Although I go in and force disable it anywhere I find it, not all users are
tech-savvy enough to know it's an issue.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JoanMCD, 2020-08-06:
That's why I wrote: "not an implementation as an Media Device !"
Here it's not more than a XML webpage - as you also can see here: xoseperez/espurna#423
(full code on µC)

It's not possible to configure something!!

cheers, Joe

starbasessd, 2020-08-06:
The difference lies in the difference between an ESP device and a full
blown PC like a Pi SBC. The exploits abound for the SBCs.
It has also been suggested to enable Avahi or similar in mEOS.
I have found that just setting up DNS to point to the router that issues
DHCP works as well, and has fewer security issues.
I am not @ccrisan or @jasaw who have also written heavily on this subject.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JoanMCD, 2020-08-06:
Simply keep it disabled by default under Services. And if its to hard to implement  - skip it. 

@starbasessd I'm not here to have a base discussion about Upnp and the would of safety problems. mEOS is not a Router where you can open ports by Upnp if you do not change the default password and also hopefully mEOS is not placed in the internet directly. I link only to the µC implementation to clarify that not the full Upnp stack is needed - only 40 lines. I don't know  >Avahi or similar< and I haven't requested it, but if this mean because of this all feature requests are blocked by default, ok.

Thx, Joe

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days

closed 2020-10-23 by starbasessd