motioneye-project/motioneyeos #2487: Cannot Get SV3C network camera to work

PaulBbecker, 2020-08-12:
In the SV3C camera’s configuration settings I set the Main Stream:
encode mode to H264, resolution 720P, Frame Rate 25, Bit Rate Type VBR, Quality Good

[image]

In motionEyeOS when I add the camera, I select “Network Camera” and enter “rtsp://192.168.xxx.xxx” as the URL. I enter the admin username and password and click OK. 

When the video comes up in motionEye it is a grey screen with a message in the upper left hand corner that says “Unable to Open Video Device”
What do I need to do to make this work?

### Preliminary Docs
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
I am running motionEyeOS version: (20200606).

### Board Model
I am using the following board/model: (Raspberry PI 4B+).

### Camera
I already have an onboard CSI camera working well as Camera1
I am having trouble adding a 2nd external camera:
I am using the following type of camera: (Network Camera).
My camera model is: (SV3C IP POE Bullet Camera).

### Network Connection
My motionEyeOS is connected to the network via: (Ethernet).

### Peripherals
I am using the following peripherals that I consider relevant to this issue:
 * (NETGEAR 16-Port Gigabit Ethernet Unmanaged PoE Switch (GS116PP) - with 16 x PoE+ @ 183W)

### Log Files
I consider the following log files relevant to this issue:
 * (attach e.g. motioneye.log)
2020-08-12 20:00:08: [motioneye]     INFO: adding new netcam camera with id 2...
2020-08-12 20:00:31: [motioneye]    ERROR: mjpg client timed out receiving data for camera 2 on port 8082
2020-08-12 20:00:51: [motioneye]    ERROR: mjpg client timed out receiving data for camera 2 on port 8082

starbasessd, 2020-08-12:
Try this page to figure out the web page to connect to:

https://www.ispyconnect.com/man.aspx?n=Sv3c

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PaulBbecker, 2020-08-14:
Thanks starbasessd! I found what I needed there. 
My camera model is B01POE-5MPL-A-2. 
Although it's not listed on that page there are some that start with B01POE-5MPL and I used their connection strings
I got connected with this address:
http://192.168.xxx.xxx/cgi-bin/video_snapshot.cgi?user=admin&pwd=123456
Note: the above username and password are the defaults 
The video is slower than the camera's built in configuration preview screen, but it's acceptable 

I also connected with
rtsp://192.168.xxx.xxx/cam1/mpeg4
However, it's responsiveness is so slow that it is unusable.  
Any ideas on why it's so slow and how to speed it up?

starbasessd, 2020-08-14:
Good to hear that you were able to connect.
Unfortunately, I have a similar issue with rtsp: on a Foscam IP camera.
I haven't been able to get it to connect real time, even though my d-link
cameras don't have the delay issue.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PaulBbecker, 2020-08-15:
Interesting about the D-Link camera working for you. On the ispyconnect site it lists most of the D-Link cameras as MJPEG. That's probably why it works so well for you. motionEyeOS supports MJPEG. I did get my SV3C camera work with the JPEG type, but it is a tad slow. It's hard to find good inexpensive outdoor POE MJPEG cameras.

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd

lucamaf, 2023-08-02:
in my case the onvif port was 854 and never manage to connect to the rstp stream