motioneye-project/motioneyeos #2488: Movies: Instead of 300 seconds, videos are written for 30 seconds

Jettochkin, 2020-08-13:
After adding more than 10 cameras - Motioneye strangely starts recording videos for 30 seconds, although in the camera settings it is set to 300 seconds.
Used Ubuntu 20.x, latest Motioneye installed via pip

### Camera
Hikvision wide 4mpx

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet. (cameras via PoE)

starbasessd, 2020-08-13:
MotionEye or MotionEyeOS?
You mention installed on Ubuntu 20.x, and installed via pip.
If MotionEye, try GoogleGroup:
https://groups.google.com/forum/#!forum/motioneye
or
https://github.com/ccrisan/motioneye/issues
It would help if you included which version is installed:

motionEye Version 0.42
Motion Version 4.2.2
OS Version Ubuntu 20.04

(from my MotionEye/Ubuntu installation)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Jettochkin, 2020-08-13:
MotionEye
Motion 4.2.2  (or 4.3.1 )

For Movies: Movie Passthrough (or h.264, or h.264/omx)

starbasessd, 2020-08-13:
https://motion-project.github.io/motion_support.html

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Jettochkin, 2020-08-17:
I'm not ready to agree that the problem is solely with Motion. I found out the reason: if communication with at least one of the cameras is lost, all cameras start sending videos for 20 or 30 seconds.

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd