motioneye-project/motioneyeos #2489: multiple issues

TongShanHui, 2020-08-13:
Hello all :)
I am having multiple issues with MotionEyeOS
A minor issue incorrect datetime is a non essential problem, but others are frustrating and very problematic.
Firstly it seems that it drops the network connection, which is wired, as wireless doesn't seem to work.
Secondly, it drops video feed, which I believe is related to the network connection.

[motion (1).log](https://github.com/ccrisan/thingos/files/5066648/motion.1.log)
[boot.log](https://github.com/ccrisan/thingos/files/5066650/boot.log)
[dmesg.log](https://github.com/ccrisan/thingos/files/5066651/dmesg.log)
[messages.log](https://github.com/ccrisan/thingos/files/5066652/messages.log)

This is running on a raspi 3B+

starbasessd, 2020-08-13:
boot.log is showing it's discovering UTC, instead of local time
motioneye is failing to start

Try re-imaging with 20200203 or 20190911
https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi3-20200203.img.xz
https://github.com/ccrisan/motioneyeos/releases/download/20190911/motioneyeos-raspberrypi3-20190911.img.xz

Make sure you put wpa_supplicant.conf in the /boot partition prior to first
boot.
Make sure you put ssh or ssh.txt in the /boot partition prior to first boot.

Let me know what result(s) you get.
Also, try hooking a monitor and keyboard to the Pi3B+ and watching the boot
process.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd