motioneye-project/motioneyeos #2494: can't read '/etc/fstab': No such file or directory

bjs-pdx, 2020-08-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606

### Board Model

I am using the following board/model: Raspberry Pi 4

### Camera

I am using the following type of camera: Multiple network cameras via RTSP

My camera model is: Primarily Wyze and Amcrest

### Network Connection

My motionEyeOS unit is connected to the network via: ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:  NONE

### Log Files

I consider the following log files relevant to this issue:
Not  sure which would be relevant but couldn't find any related information in any of the logs.

I'm running  motionEyeOS on a Raspberry Pi 4 and am trying to create a mount point for an NFS share on my NAS device.  I read the post at https://github.com/ccrisan/thingos/wiki/fstab.user and created an fstab.user file in /data/etc which included the mount data for the NAS.  I also created a mount point in /data.  When I try to mount the share I receive the following error message:  "mount: can't read '/etc/fstab': No such file or directory" which indicates to me the system isn't reading the fstab.user file in /data/etc.  Rebooted the system just to see if it only reads the file on startup without success.

What am I missing?

bjs-pdx, 2020-08-24:
Ran the following command manually:  mount -T /data/etc/fstab.user -a and got the response:  failed: Connection refused.  On my NAS I get:  Aug 24 04:39:44 freenas1 mountd[82418]: mount request succeeded from (ip address of MotioneyeOS system).  

No files on mount point...

starbasessd, 2020-08-24:
Under Expert Settings, If you turn 'Debug' on, it'll reboot, and then you
can edit the file system as needed to enable NFS. (/etc/fstab)
When done, you need to turn 'Debug' back off.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bjs-pdx, 2020-08-25:
The Pi reads the fstab file an tries to mount the share but I get:
mount: mounting 192.x.x.x:/mnt/SSD/Motion on /data/motion failed: Connection refused

However, my NAS device says that the request was completed:
Aug 24 22:43:28 freenas1 mountd[82418]: mount request succeeded from 192.x.x.y for /mnt/SSD/Motion

I have the same settings for this share on my ubuntu laptop and several other systems which access the share without problems.  Which log file on the MotionEyeOS system might shed some light on the problem?

bjs-pdx, 2020-08-25:
Silly question--does motioneyeos include the nfs utilities in the installation, specifically nfs-common?

bjs-pdx, 2020-08-25:
Finally found this:  http://www.armadeus.org/wiki/index.php?title=Network_File_System_configuration#Connection_refused

This allowed mounting of the file share:

Connection refused

    If you get this on your APF when trying to mount:

mount: mounting 192.168.0.2:/local/export on /mnt/nfs/ failed: Connection refused

    then your kernel might need some extra options:

mount -o port=2049,nolock,proto=tcp 192.168.0.2:/local/export /mnt/nfs/

bjs-pdx, 2020-08-25:
Somewhat frustrated by this.  Created fstab file using debug mode in /etc containing the following entry:
  192.x.x.x:/mnt/SSD/Motion /data/motion nfs rsize=8192,wsize=8192,timeo=14,intr,nolock,proto=tcp,port=2049 0 0

When I reboot the mount fails.  If I run "mount /data/motion"  the mount completes and I have file access.  I tried the same entry in /data/etc/fstab.user which also didn't mount at boot.  Network timing issue?

starbasessd, 2020-10-23:
Did you figure it out? 
If it's a timing issue, you could add a cron job to auto run a script @boot, in the script you could have a delay (sleep x)(seconds) before running the mount command.
 Let me know either way.

bjs-pdx, 2020-10-23:
I was never able to get it working.  Just gave up for now.  Too many things going on.

starbasessd, 2020-10-23:
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd