motioneye-project/motioneyeos #2495: Low framerate RAspberry Pi3b+

LTCD474, 2020-08-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190728.

### Board Model

I am using the following board/model: Raspberry PI 3B+.

### Camera

I am using the following type of camera: RaspberryPiCam, Logtech C930e, Lolgitech 720p, Microsoft Cam 720

### Network Connection

My motionEyeOS unit is connected to the network via: LAN 1Gbit.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Rapberry, RaspbiCam and Webcams only

### Log Files

I consider the following log files relevant to this issue:

cant upload them right now

Hello!
At first! Thank you for this nice Software!

Well i got some Problems to get constant framerates and i got big delays.
System is used for a model train project to observe the trains moving below the dioramas (well you would need xray eyes to do so ;) ).
I use 3 Raspberrys with RaspberryPiCam Module (i think V2) and some other Logitech webcams over USB.
On the computer (Windows) via the browser (used Chrome, Firefox, Opera, Seamonkey, Brave) I access the Raspberrys to get the Vidoestream.
To my Problem.
When i use raspbian with the video commands, the videos are smooth, about 30-50fps and there was nearly no delay.
But with motioneye OS, I had a big delay (about 10-20 seconds) and the video is stuttering and gets only 5-15fps (even with 15fps, the stream was just laggy)

I am sorry that i cant send the logfile right now, but i didnt change anything on the system or OS.
(Nearly) Everything is deactivated, no motion detection, no streaming. no recording, no still image or whatever.
Every Raspberry got a static IP and is connected via LAN in an 1GB/s network (no internet, only the 3 Raspberrys and the Computer is connected)
(Computer is prepared to get 16 Monitor Outputs)
Every Cam is set to 800x600 and 30fps in the settings

First the Raspberrys got about 60-68°C, when we installed an active cooler, they went down to 43°C (maybe i will install heatsinks too)

With everything deactivated, the delay is just 1 second, that Problem is I think fixed.
But if I access the Raspberrys (1 Raspberry is using just the RaspberryPiCam and nothing else) with the browser and a train is passing by, the Video is just shown like a Diashow/slideshow.
Did i miss something to configure or are there commands to speed it up?
I get even the same result with just 1 Pi and 1 Cam on lowest settings or with all and on HD (1920x1080) settings.
Are there some tweaks for the command box? (hope thats the correct designation)

Thank you in advance
Regards
Chris

starbasessd, 2020-08-24:
You can speed up some by setting the PiCam Pis to 'Fast Network Cam', and
on the Hub Pi you can try to set the Video to 'Pass Through'.
If you don't need recording, you can disable Stills, Movies, Motion
Detection & Notifications.
I have currently 5 cameras of various types, and the best I get across the
network is in the 5 FPS per camera, the limitations being the network
(Ethernet) speed caused by the USB2 limitations. Wifi is even slower, (3-5
FPS)
All my cams are set to 800x600...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

LTCD474, 2020-08-26:
Thank you for the advise.

I will try to upload tomorrow or Friday the LogFile etc
(Pis not at my home)

Everything is disabled, well i enabled Video Streaming again, cus its easier to access all the other connected cams.
I set the fps rate in the Ratio to 20, 25 and then 30 fps, also in Videostream (Quality from 55to75%) but still not that success and its slow :(
(well i wont reach the 30fps thats shown in the settings)

so stay tuned for the files

starbasessd, 2020-10-23:
Status?
You will get better results with newer versions, too.