motioneye-project/motioneyeos #2496: Date/Time acquisition failing on boot on Raspberry4

RedHart66, 2020-08-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

latest version: 20190911
kernel: 4.19 (raspbian)

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).
Raspberry PI 4 Model B 4G Ram
### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

Raspberry Pi Camera Module V2 

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).
Wi-fi
### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)
USB WI-FI adapter plus wireless keyboard
### Log Files

I consider the following log files relevant to this issue:

Cannot get logs out of raspberry pi onto my pc or Mac with motioneyeos.

When using ntp for date accessing Message is:
Exiting name server cannot be used:Temporary failure in name resolution(-3) 
[used ntp1.npl.co.uk]
When using http and google.com message is : 
failed
Starting http date updater:done

This is similar to #2493 but on Rasberry4. Workaround for #2493 , use of earlier release was for Raspberry3
Which is presumably no good for Raspberry4.
The Raspberry Pi4 OS boots ok and finds the correct date and time.

Please advise!

starbasessd, 2020-08-25:
How are you setting up your wifi?
If you install RaspberrfyPiOS and set up your wifi the same way, can you
ping pool.ntp.org?
It sounds like your DNS server or gateway aren't set up correctly.
If you can connect a monitor to the Pi4, does it show any issues?
The latest releases are 20200203, & 20200606 found here:
https://github.com/ccrisan/motioneyeos/releases

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RedHart66, 2020-08-26:
I am using wpa_supplicant with the name and password of my router.
I am using a static address for motioneyeos.
The boot log says:
Starting wpa_supplicant: done (brcmfmac)
Configuring wireless network: <My static address>
Etc.

I can ping motioneyeos from my pc.
I cannot ping pool.ntp.org or google.com from motioneyeos, get bad address pool.ntp.org.
If install the Raspberry OS, I can ping pool.ntp.org

I shall try the latest builds.

Thanks for the swift response!

starbasessd, 2020-08-26:
How are you setting the static address

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RedHart66, 2020-08-26:
I used the Network menu in motioneyeos.
The boot messages say it has the correct address.
I have no Linux server, is there a way to get a file out of motioneyeos like the
Boot file. Whenever I try a command it is missing. As a very old unix programmer from 35 years ago it is all bit hazy now what commands I could use.
Thanks again

RedHart66, 2020-08-26:
Hit wrong button!

starbasessd, 2020-08-26:
Several ways to get the log files.
Easiest may be to use a Live Linux (like Ubuntu) USB thumb drive, boot a PC
to it, then use whatever you used to create the SDCard to read it.
As to setting the static IP, I use my DHCP server to always give my Pis the
same address.
You can create a file called static_ip.conf and place it in the /boot
folder before first boot.
In that file, include the following lines:
---
STATIC_IP="192.168.42.1/24"
STATIC_GW="192.168.42.1"
STATIC_DNS="192.168.42.1"
---
(use your network numbers)
But, be sure to not touch the network settings afterwards in the WebGUI, as
it will break things.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RedHart66, 2020-08-27:
The problem of date failure seems to be related to the creation of a static address. If this is not done 20200606 boots and gets the date correctly. If a static address is set up via the web gui the next boot fails to get the date.
Adding static_ip.conf file produces the same result. My static-ip.conf file is:
STATIC_IP=“192.1.168.200/24”
STATIC_GW=“192.1.168.200”
STATIC_DNS=“192.1.168.200”

My router uses dhcp addresses 192.1.168.64 to 192.1.168.254.
I had tried address 50 as well (out of the dhcp range)
Any suggestions?

Thanks!

starbasessd, 2020-08-27:
GW and DNS need to point to your Router, not to itself.Most routers are
either .1 or.254

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-08-27:
(Sorry, accidentally hit send) It would appear that your router is sitting
on .1. So change your static_ip.conf to:

STATIC_IP=“192.1.168.200/24”
STATIC_GW=“192.1.168.1”
STATIC_DNS=“192.1.168.1”

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-08-27:
Also, not a good idea to use 192.1.168.x for internal addresses.
192.1.168.x is internet routable.
try using 192.168.1.x/24 or 172.16.x.x/16 or 10.x.x.x/8 are all by design
used for internal IPv4 address subnets, because Internet Routers won't
forward them.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RedHart66, 2020-08-27:
Sorry I gave you some wrong information, I typed the addresses in wrongly. My static config file now is:
STATIC_IP="192.168.1.200/24"
STATIC_GW="192.168.254"
STATIC_DNS="192.168.254"
My router is on 254
[I am trying this simpler one before your second suggestion, which I will use]
Using this the OS fails to boot, keeps retrying.
Configuring wireless network 192.168.1.200/24
...
Setting static gateway to to 192.168.254: ip: RTNETLINK answers Network is unreachable
failed
Setting static DNS server to 192.168.254:done

Will use of the subnet addresses help this?
Thanks!

starbasessd, 2020-08-27:
I'm hoping you are still typing it wrong.
BTW, 254 is in your DHCP range, and your router should be on a static.
Boot it DHCP, and tell me what
ifconfig | grep inet <enter>
command reports on the PI
Then tell me what
route -n <enter>
reports
This will tell me (you) what you should be setting in static_ip.conf for GW
and DNS

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RedHart66, 2020-08-27:
Ifconfig yields:
Inet 127.0.0.1 netmask 255.0.0.0
Inet6 ::1 prefixlen 128 scoped 0x10<host>
Inet 192.168.1.73 netmask 255.255.255.0 broadcast 192.168.1.255
Inet6 fe80::dea6:32ff:fe7a:2c16 prefixlen 64 scopeid 0x20<link>

Route -n yields:
Kernel IP Routing Table
Destination Gateway          Genmask.              Flags.  Metric. Ref.  Lface
0.0.0.0.         192.168.1.254. 0.0.0.0.                  UG.        0.       0.     Wlan0
192.168.1.0.   0.0.0.0.            255.255.255.0.     U.          0.        0.     wlan0

RedHart66, 2020-08-27:
I have tried that and used Angry scanner to find unused addresses. Pinging them reports they are unreachable. 

I think I have to work round this now and avoid static addresses . You have been a great help but I can’t keep bothering
you. Your swift responses have been great.

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd