motioneye-project/motioneyeos #2497: PROBLEM: Pine64 - Upon first boot, Unwritable files (READ ONLY file system)

frolleswe, 2020-08-26:
###motionEyeOS Version
I am running motionEyeOS version: motioneyeos-pine64-20200606
###Board Model
I am using the following board/model: Pine64
###Camera
My camera model is: Logitech C270
###Network Connection
My motionEyeOS unit is connected to the network via: Ethernet, trying to make a wifi link by following the WIFI instructions but with no success. 
###Logs
Cannot get logs out of Pine64 onto my pc with motioneyeos.
###ISSUE
I have an issue with my Pine64 trying to boot the image.
Upon booting it says 
```
Starting motioneye: 
*Staring smbd: failed
mkdir: can't create directory 'var/log/samba': read-only file system
mkdir: can't create diectory '/var/lib/samba/': read-only file system
*starting nmbd: mkdir failed on directory /var/log/samba/cores: no such file or directory
Failed to create /var/log/samba/cores for user 0 with mode 0700
Unable to setup corepath for nmbd: No such file or directory
failed to configure logging: [Errno 30] Read-only file system '/var/log/motioneye.log'
```
System keeps on rebooting
How do I go on? Please advise!

starbasessd, 2020-08-26:
Have you tried 20200203, or 20190911?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

frolleswe, 2020-08-26:
Thanks for the tip @starbasessd , I will return with an answer if it works 👍

frolleswe, 2020-08-26:
I can confirm that the motioneyeos-pine64-20190911.img version works. Thanks for the help @starbasessd

closed 2020-08-26 by frolleswe

starbasessd, 2020-08-26:
Good deal, glad to help.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?