motioneye-project/motioneyeos #2499: Mask privacy

titof2375, 2020-08-29:
hello, i have set up a motion picture that works perfectly, thanks to you.  when, i read what motion does, it is marked privacy mark that cannot be found.  we can make an editable mack my privacy mask steps

starbasessd, 2020-08-29:
*n motionEyeOS, under Motion Detection, Turn Mask ON*

*It will then show you options for the Mask.*

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2020-08-29:
so when I activate the mask, I only have 2, Mask smart and mask editable

starbasessd, 2020-08-29:
Select Mask Editable

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2020-08-29:
yes, on the other hand I still see the sectioned party

starbasessd, 2020-08-29:
To physically (optically) block it out would require a cut-out piece of
paper or plastic covering the lens.
The processing power to block specific parts of the electronic image would
put it out of the reach of most SBCs.
motion/motionEye doesn't support that function, as far as I am aware.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2020-08-29:
it's in the doc
[image]

starbasessd, 2020-08-29:
Don't know, then.
Good luck.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Privacy Masking is enabled in Settings, Video Device, Privacy Masking in motionEye and MotionEyeOS as of 20200606 and later.

closed 2020-10-23 by starbasessd