motioneye-project/motioneyeos #2500: Constant Freezing

lonewolph, 2020-08-29:
My Cameras Constantly Freeze up. Sometimes they will work for days or even weeks, but even more now lately they are frozen by the next day. Could anyone tell me why?

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 4.2.2+gitUNKNOWN

### Board Model

I am using the following board/model: Raspberry Pi 3 B+

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your ca
[dmesg (1).log](https://github.com/ccrisan/motioneyeos/files/5146072/dmesg.1.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5146073/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5146074/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5146075/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5146076/motion.log)

mera model, e.g. Logitech C290).
3 Cameras:
#1 Logitech C270

#2 https://www.amazon.com/gp/product/B00VFLWOC0/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

#3 https://www.amazon.com/gp/product/B07D8S6JRH/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).
Ethernet
### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-08-29:
> dt_b_search_asin_title?ie=UTF8&psc=1

Which version of motionEyeOS?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lonewolph, 2020-08-29:
motionEye Version | 0.42.1
-- | --
Motion Version | 4.2.2+gitUNKNOWN

starbasessd, 2020-08-29:
20200606, 20200203, 20190911, or other?

lonewolph, 2020-08-29:
OS Version | motionEyeOS 20200606
-- | --

starbasessd, 2020-08-30:
Try 20190911 or 20200203

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lonewolph, 2020-08-30:
I will give that a try, thank you

starbasessd, 2020-10-23:
Status?

lonewolph, 2020-10-23:
I noticed an improvement but hasn't eliminated the problem

starbasessd, 2020-10-23:
Have you tried one of the latest dev- versions?
If you can ssh into the Pi or use the console, what do you get for the following:
vcgencmd measure_temp [enter]
vcgencmd get_throttled [enter]
preferably after a camera freezes. I am assuming that the stream from the cameras is still working, and just not showing on the Pi streams.