motioneye-project/motioneyeos #2502: Odroid with 20190911 in boot loop with green screen

lx-zander, 2020-08-31:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (20190911).

### Board Model

I am using the following board/model: odroid c2.

### Camera

I am using the following type of camera: (not relevant but HDR USB webcam 

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (original odroid wifi stick).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
- original Odroid wifi stick
- rapoo mouse and keyboard
- HDR webcam / logitech 720p cam

### Log Files

cant grab any log files.
 
### Issue:

after running motioneye os perfectly for 1 month, i did an update, OS
asked me for.
After this server was crashed and hang in a boot loop, with green screen.
Blue led is flushing normal for a minute then off then constant for 10 sec,
then starts flushing again... Red LED is always on.

I did try: new emmc card, new power supply, new odroid, checked odroid with
other OS (ubuntu mate- workes fine), flashing new (20190911 OS), booting with 
and without peripherals.

Flashing emmc with etcher on win10, works fine??? no errors, filesystem visible
Want to try old version of motionEyeOS, but coudnt find it on git.
Please help!

[image]

starbasessd, 2020-08-31:
Older releases can be found here:

https://github.com/ccrisan/motioneyeos/releases

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lx-zander, 2020-08-31:
Thanks.
I tried: motioneyeos-odroidc2-20200606, motioneyeos-odroidc2-20190413.img, motioneyeos-odroidc2-20190911.img
with etcher and Win32DiskImager, for some times
Same result- green screen and boot loop...
No idea, what i made wrong...
zander

starbasessd, 2020-08-31:
I do not have an Odroid, so I can't help you further. Maybe another user
can assist.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lx-zander, 2020-08-31:
Many thanks so far!!!

starbasessd, 2020-08-31:
Oh, can you connect to ethernet and try booting, or a different wifi
dongle? On the RPis, if it cannot find and set up networking, it'll give
you the reboot system.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lx-zander, 2020-08-31:
Sure. Wifi dongle is original odroid, that worked perfekt before.
Ethernet connection has the same effect. Green screen- reboot.
Will try to pre-configure the wifi.

ccrisan, 2020-08-31:
Just a heads-up: Odroid C1/C1+ and C2 will be removed from the supported boards starting with the next release. The reason behind this decision is the fact that they appear to be abandoned boards with no mainline kernel support and no decent effort of running a relatively recent kernel (4.x).

lx-zander, 2020-08-31:
Ohhh no! This info killed my day by far.... this the superlative of frustration. Sorry but thanks...
zander

lx-zander, 2020-09-01:
hy i did managed to connect the broken system by ethernet cable. With advanced ip scanner i coudt establish a connection via console but have no idea if this helps. No files there and can not find log file.
[image]

starbasessd, 2020-09-01:
what does
pwd<enter>
give you?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lx-zander, 2020-09-01:
not that much...
[image]

starbasessd, 2020-09-01:
OK it just means by default you are in the home folder for root.

Now, try either
ifconfig<enter>
or
ipconfig<enter>
These will give you the ethernet and wifi settings

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lx-zander, 2020-09-02:
hello,
i think i stop this and going to get a raspi... hopefully more luck. Thanks a lot!!!!

starbasessd, 2020-09-02:
Gotcha. Don't forget to close the ticket.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lx-zander, 2020-09-02:
Sorry- forgot to check the #ifconfig command.
There was good outcome- no??                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           
[image]

starbasessd, 2020-09-02:
Just shows you are getting the ethernet connection.
Doesn't show any wifi connection.
Without an odroid here, it's hard for me to know what is broken.
And @ccrisan is discontinuing support/builds.
I might get an odroid but it wouldn't be a C2...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lx-zander, 2020-09-02:
May be last idea. This is the root folder of the emmc to boot. Is this locking ok? Shoudnt be there folders, instead of image??? I tried to preconfigure the wifi, but here is no folder boot to put the file in...
z.
[image]

starbasessd, 2020-09-02:
No idea, sorry.
Normal linux has a bunch of folders off of /, including bin sbin var root
home
 etc...
Dunno about odroid.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lx-zander, 2020-09-02:
Yeah this is different. Checked in odroid forum- but too time consuming
for me. But its a pitty- pretty good machine for such small jobs.
Just to tell you- i checked the emmc of running uubuntu on both
mashine- looked the same... some files/ image file/ no folders.

Really thanks a lot. If i get mad- i will try by myself. Cant stay to have cool small
machines hanging around- nothing to do! Maybe a good gift for somebody else....

thanks a lot again. Was nice experiance with the OS and your support!!!
Have good live.
See you when i see you!
zander

closed 2020-09-02 by lx-zander