motioneye-project/motioneyeos #2503: Google Drive Issue

u4h37u, 2020-09-03:
Hi,

My motioneyeOS system has been running for three years without issue.  However, a few months ago I noticed that the google drive uploads were only being retained for one day before being deleted.  I had the clean cloud setting enabled but the retention policy for local files is one month for movies and one year for still images.

To try and fix the initial issue, I simply turned off the clean cloud setting.  Google drive was then happily storing data older than a day.

However, on the 11th August, no new files seemed to upload to google drive.

To try and resolve this myself, I clicked the "test service" link in the file storage section.  This gave me a spinning wheel which did not go away.  I then renewed the authorization key to see if that would help but it has not.  I don't know where to look next.

Does anyone have any troubleshooting tips?

Thank you!

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi 3 Model B Rev 1.2

### Camera

I am using the following type of camera:  Network Camera

My camera model is: SV3C POE Camera

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

External hard disk

### Log Files

I consider the following log files relevant to this issue:

motioneye.log
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5171048/motioneye.log)

starbasessd, 2020-09-03:
When did you upgrade to 20200606? Would the issues have started  near or
right after the upgrade?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

u4h37u, 2020-09-03:
Hi, Thanks for the quick response.  I can't say for sure, I'm sorry, but I think I did upgrade to the latest version to try and fix the issue so I don't think the issue was caused by the upgrade.

starbasessd, 2020-09-03:
Can you try either downgrade to 20190911 or upgrade to 20200831, both of
which are found here:
https://github.com/ccrisan/motioneyeos/releases
Best to do a backup prior to re-image, then restore to recover most of your
settings.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Status?

starbasessd, 2020-11-05:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-11-05 by starbasessd