motioneye-project/motioneyeos #2504: bananapi m1 error?

hs6666, 2020-09-04:
Hi,

I have a bananapi happily running an old version of motioneyeos, trying to upgrade....

I downloaded (https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-bananapim1-20200606.img.xz) and burned to a sdcard using w32diskimager ... no errors

at boot up keep seeing various "read only filesystem" errors during initial boot up

it does show it getting an IP address, then holds on trying to correct time offset, then just reboots and keeps doing that

have tried different sdcards and burned using different PC's on different broadband connections, but same error

any ideas please?

any ideas please

starbasessd, 2020-09-04:
Try 20200203 (prefered), and dev20200831 found here:
https://github.com/ccrisan/motioneyeos/releases
and report your results, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

hs6666, 2020-09-04:
Hi, afraid the same.... some screenshots from first run attached if it helps, ta..

[image]
[image]
[image]

starbasessd, 2020-09-04:
I don't know. It's another board I don't have access to.
For the RPis, it would indicate a corrupted/damaged SDCard or downloaded
image where the file system is read only on first boot.
I would assume that you've tried the normal, full-blown OS for the BananaPi
(not motionEyeOS) to see if it still boots normally, is that true?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-09-04:
> Hi, I've just burnt the backup image I created on the same sdcard on the
> same bpi, and it's working fine, its...
>
> motionEye Version | 0.39.2
> Motion Version | 4.1.1
> OS Version | motionEyeOS 20180627
>
> It works perfectly, six camera setup, one local USB, sata drive attached
> :) .... accept its such an old image now it won't work with some browsers
> properly
>
> Can I do upgrade via ssh somehow, I know my way around debian :)

You can't directly update via the Update button as there have been too many
changes to the structure of the system (if you look at the releases since
then many say you cannot upgrade from earlier versions.
The motionEyeOS builds use Buildroot to generate the systems, which are
very locked down. Things like apt, dpkg etc are not included.
For me, it would come down to 'if it ain't broke, don't fix it', at least
until @ccrisan can look at it. I don't know if he has a board to test on,
or if anyone else would be willing to check out the newer images.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

hs6666, 2020-09-04:
Hi, bit of an update after reading your comment above.....

managed to update to 20181209 after burning the image and restoring form backup

then managed to upgrade to 20190119 by following the instructions at https://github.com/ccrisan/motioneyeos/wiki/Upgrading-From-Versions-Prior-To-20190119

So I am now on the new partition layout :)

But..... for some reason I still cannot upgrade to the latest version... even though update now shows that a new version is available.... when I click update, it says upgrade is successful... but its not true, I'm still on 20190119 ??

Have tried to upgrade to the next build via ssh by issuing the command "fwupdate upgrade 20190427" all the way up to "fwupdate upgrade 20200606", but each time I get the following......

upgrading to 20190427
downloading...
downloaded 20190427
extracting...
losetup: /data/.fwupdate/firmware.img: No such file or directory

I suppose its progress :)

starbasessd, 2020-09-04:
2 things:
Check free space (need several (>3GB) free space.
Try saving a backup, and do another manual update to 20100427, or a newer
version. then restore.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

hs6666, 2020-09-04:
oh ok that explains it as it's only a 4GB sdcard

If i was to make a backup of this image and burn it to a 8GB sdcard, how would I go about expanding partition mmcblk0p3 into the freespace on the new sdcard?

PS currently "fdisk -l" is giving me....

/dev/mmcblk0p1         2048   63487   61440   30M  c W95 FAT32 (LBA)
/dev/mmcblk0p2       204800  614399  409600  200M 83 Linux
/dev/mmcblk0p3      2097152 7741439 5644288  2.7G 83 Linux

starbasessd, 2020-09-04:
What does the  df -h  command show? (in a ssh session, or at the console)
You can try to copy it, but better to do the backup  (in the Admin WebGUI
under 'General') / restore (image the 8GB with the newer image, and do the
restore from the same location)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Status?

starbasessd, 2020-11-05:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-11-05 by starbasessd

Tey, 2022-10-20:
I had a similar issue with bananapi m1 and v20200606. It seems this version will not switch the root partition to read-write, and since it is not mounted as rw in the boot arguments, it stays in read-only mode.

I fixed that by adding the "rw" argument to the bootargs variable in [the boot script](https://github.com/Tey/file_drop/blob/main/motioneyeos/bananapi/boot.cmd). The binary version of the script can be downloaded [here](https://github.com/Tey/file_drop/raw/main/motioneyeos/bananapi/boot.scr), just replace the "boot.scr" file in the FAT32 partition with this one after creating the SD card image, and the system will boot without any issue afterwards.