motioneye-project/motioneyeos #2505: increasing list_media_timeout

tstudebaker, 2020-09-06:
### Preliminary Docs
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
20200203

### Board Model
Raspberry PI 3B+ and Zero.

### Camera
Raspberry Camera v2.1

### Network Connection
WiFi

### Peripherals
USB-HDD

My camera system runs ok, but sometimes there are too much pictures taken so that I get the error:
"Failed to get picture list for http://xxx.xxx.xxx.xxx/config/1: request timed out".

Searched the forums and found a discussion "Motieneye/#647: Unable to view media".
There was a suggestion to increase the value of the "list_media_timeout"-entry (default value 120) in the motioneye.conf-file.

Now, I am using MotioneyeOS, but found two motioneye.conf-files in directories "./data/etc/" and "./etc".
None of them holds the "list_media_timeout"-entry so I guess the system uses a value of  20 which generates timeout after about 105 seconds.

Is it ok to add the "list_media_timeout"-entry with a value for example 240 in the motioneye.conf-file?
Which motioneye.conf file should I edit??

starbasessd, 2020-09-06:
I do so, and added it to the one in /data/etc/motioneye/.
The one in /etc/motioneye/ is read-only, unless you turn on Debug.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tstudebaker, 2020-09-06:
I did append the "list_media_timeout 360" in the /data/etc/motioneye.conf of the Rpi3B+Hub and the 2 Zero-Satellites.

[image]

It also seems to me that I do not have a directory "/data/etc/motioneye/".

[image]

After appending I booted all 3 devices, but still the list-function gives timeout error after about 105 seconds.
At the moment the error appears on one of the satellites, hub and the other satellite do the listing just fine.
No matter if I try to list directly from the satellite or via the hub.

All the pictures from the hub-camera and the 2 satellites are stored in the USB-HDD, which is attached to the hub.

Any ideas ?

starbasessd, 2020-09-06:
Will check in a bit, working another issue right now.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-09-06:
Sorry, typo on my part.
Try the /data/etc/motioneye.conf file
How do you have the 2 PiZero's connected? As Remote motionEye, Simple
mjpeg, o Network?
I have all mine connecting as Simple mjpeg, so that all processing is
handled by the Hub, rather than on both a Zero and the Hub. Saves
duplication of effort.
I have Zeros with motion detection, snapshots and pictures off, only
streaming on.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tstudebaker, 2020-09-06:
>>How do you have the 2 PiZero's connected?
I am not sure I understand the question but:
the 2 Zeros and the Hub have motionEyeOS installed, and the cameras of the Zeros have been added to the Hub.
The Hub also has its own camera.
This is how the Video Device of one Zero is seen by the Hub:

[image]

The same Video Device seen by the Zero where it is attached:

[image]

The Zeros and the Hub are connected together by the wifi created by my router.
My Zeros (and the Hub) have ON: Video Devices, Text Overlay, Still Images and Motion Detection.

My goal is that the zeros (and also the hub) figure out (by motion and/or pixel change) if a picture should be taken, if it is taken, it is saved in the USB-HDD which is attached to the Hub.

starbasessd, 2020-09-06:
My Zeros are Wifi Connected.
My Zeros are set to Still Images Off, Movies Off, Motion Detection Off. (On
the one with a CSI MMAL Camera it is set to Fast Network Camera ON)
My Hub (with a USB Camera) does  all the recording, Still, Movie, Motion
Detection are on. I save to a NAS but it would be the same if it was to
local USB.

If you are using the USB attached storage on the hub for the PiZeros
directly (Like a NAS) and the hub is also recording the Zeros to the USB
storage, then you are getting duplicates on the USB storage?

Zero (recording to hub USB) _____
                                                        |---- Hub
(recording to USB stick)
Zero (recording to hub USB)---------|

If both are using MMAL (CSI) PiCams, I'd set the Zeros to Fast Network
Camera ON and so all the heavy lifting on the Hub. It can determine if
motion on all the cameras and record as needed.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tstudebaker, 2020-09-06:
OK, I will test that kind of setup, but how does our discussion relate to the original question of increasing "list_media_timeout", and the observation that editing "/data/etc/motioneye.conf" did not have any effect on it ?

starbasessd, 2020-09-06:
You would then only have 1 location to have to send notification from (The
Pi3 hub) and the setting is in /data/etc/motioneye.conf on that PI. BTW,
even the higher timeout may not be sufficient if you have a LOT of images.
You may want to move the files from the primary path to a 'saved' path,
maybe as often as daily, depending on how many pics are being grabbed
hourly or daily.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tstudebaker, 2020-09-06:
Ok, that makes sense, I will report after I have done some testing.

tstudebaker, 2020-09-07:
Changed the other Zero/Satellite to Fast Network Camera, but that changed the camera views from the hub totally so let's consider that another time.

But the file storage system has totally confused me now.

I have defined the File Storage of the Zero/Satellite (goal: Zero/Satellite saves its pictures only in the hub/USB-HDD) as:

[image]

And in the Hub (goal: the hub saves its pictures only in the USB-HDD attached to it) as:

[image]

I have created this directory structure in the hub/USB-HDD (explained in issue#2470):

[image]

To my surprise I found (instigated by our discussion) also this directory structure in the hub/USB-HDD:

[image]

If I connect directly to the Zero/Satellite it shows this directory structure:

[image]

My original goal was that by defining the File Storage of the Zero/Satellite (Garage) as expressed in a picture above would make it save the files only in the directory I created: "/home/ftp/storage/sda1/Cameras/Garage", but clearly this is not the case.

Are the pictures of the Zero/Satellite now in three different places, 2 being in the Hub/USB-HDD and 1 in the Zero/Satellite itself ?
What should I do to make the Zero/Satellite save its pictures only in the Hub/USB-HDD ??

starbasessd, 2020-10-23:
On the Hub, turn on in Services, FTP Write Enable and SAMBA Write Enable.
On the PiZero's you would enable File Storage,
Storage Device Network Share
Network Server <ip address of hub>
SMB Protocol Version 1.0
Share Name storage
Share Username admin
Share Password <blank> or whatever you set it to
Root Directory /
Apply at the top BEFORE clicking Test Share
This will force all files created on the PiZero to save only to the hub.
If you want them on the PiZero AND the hub, let me know, it's a different setup.