motioneye-project/motioneyeos #2506: Pi 4 WPA_Supplicant "Failed" - even after configuring over ethernet.

dragonpme, 2020-09-06:
Pi 4 WPA_Supplicant "Failed" - even after configuring over ethernet.

I am running motionEyeOS version: motioneyeos-raspberrypi4-20190911.img.xz
I am using the following board/model: Pi4 4GB.

starbasessd, 2020-09-06:
Interesting.
Can you try 20200203, or dev20200831?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JohanD78, 2020-09-06:
Can confirm partly confirm and deny.

I have this problem with devbuild 20200831 but it works fine with all other builds on the pi4.

With 20200831, if i start clean with ethernet it works fine, but wont boot if i add wpa_supplicant.conf (even with ethernet plugged in)

Same if i config wireless when plugged in over eth, the pi the starts to bootloop due to no connection.

2.4GHZ or 5GHZ dont matter, same result.

Pi4 4Gb here aswell btw.

To make it "worse": My pi3b´s both work fine wirelessly with 20200831 and since i cant reach the pi4 i cant provide any logs ;/

Sorry for hijack, but tought it worth to input what i know.

starbasessd, 2020-09-06:
I have confirmed in another thread.
I was able to post logs messages, boot.log, and dmesg.log in that thread.
It's as if the driver files for the Pi4wifi chip aren't included. Being a
BuildRoot build, (and I have only one level of luck with that - NONE) I am
not able to modify the /boot folder to include necessary files (it uses
init.rd).
Reported and flagged to @ccrisan.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Closed as duplicate.

closed 2020-10-23 by starbasessd