motioneye-project/motioneyeos #2507: How can I watch WI-FI cameras encrypted on MotionEyeOS?

ivanjf1978, 2020-09-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-09-07:
Simple answer: You Can't.
More complex answer: You need to answer the questions, especially about the
make and model of the camera.
Cameras like RING and NEST and similar run a proprietary encryption that
motionEye/motionEyeOS doesn't support.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2020-09-08:
Hi @starbasessd ,

Thanks for replied to me. 

The camera it is this one:

https://www.amazon.co.uk/gp/product/B089DDXV1W/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

Model: Littlelf 1080P Wifi Camera, Security Camera

I can not find the URL. I found the ip address. but, I can not find the URL.

Any advice how can I access with MotionEyeOS. Any other options for use this camera on MotionEyeOS?

Thank you very much

Regards

Ivan

starbasessd, 2020-09-08:
I did a quick search, and they are like RING and NEST. Cloud only access,
proprietary encryption. MotionEye/motionEyeOS doesn't support them.
No rtsp://, http:// mjpeg, or stream access that I could find.
Most of the data was from 2017, and I don't think they are still a viable
company.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2020-09-08:
@starbasessd thank you very much for your help

ivanjf1978, 2020-09-22:
I have tried to add the ESP32 Cam in MotionEyesOS and not working.

This is the camera that I tried to add, any advice please? Please say it is possible!!!

https://www.ebay.co.uk/itm/1X-Esp32-Cam-Camera-Wifi-Bluetooth-Module-4M-Psram-Dual-Core-32-Bit-Cpu-De-W2E/224112725792?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

Regards

Ivan

starbasessd, 2020-09-22:
I do not see any links to manuals or output specifications. If it supports
jpeg/mjpeg or RTSP:// out put, it probably could.
Do you have a link to any kind of manual/programming guide?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cap9qd, 2020-09-22:
You will have to load software on the esp32 to setup and output either rtsp
or simple jpg. I managed to make it work but nothing more than a proof of
concept (ie I don't use it for my regular cameras) . If you search around
for rtsp and esp you will find it on github... I don't have it at hand now.

ivanjf1978, 2020-09-22:
Thank you very much @starbasessd and @cap9qd for replied. Sorry I am not expert in this, I am only watch tutorials on youtube and try to do something like surveillance system for my Raspberry Pi 4B with MotionEyeOS. 

I am not sure about if the board ESP32 cam supports jpeg/mjpeg or RTSP:// out put. I think it could support. But, I don't know how check this. The only thing that I know is, I can see the camera with IP address (URL) after the programing in Arduino IDE on google chrome, but when I put the URL address on MotionEyeOS it says "it can not display the video"

I have found this tutorials about the ESP32 board that could be useful to find out if the board supports jpeg/mjpeg or RTSP:// out put, but I can not find the answers yet.

https://randomnerdtutorials.com/esp32-digital-inputs-outputs-arduino/

https://learn.circuit.rocks/esp32-cam-with-rtsp-video-streaming

I have tried to make this project and see the stream video on MotionEyeOS and record the video in my google drive

https://www.youtube.com/watch?v=ywCL9zDIrT0

I am looking for that, as soon I will find it I will let you know

Thanks

Regards

ivanjf1978, 2020-09-28:
Hi  @starbasessd, I can not find the answer if the ESP32-Cam supports
jpeg/mjpeg or RTSP:// out put. Please any help?
Thanks

starbasessd, 2020-09-28:
Unfortunately, no.
I do not have any ESP cameras, and the lack of online documentation for
that model doesn't help.
I do not have current resources to purchase one for experimentation or
support.
@cap9qd stated he had it working as 'proof of concept', but nothing further.
I'm sorry.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2020-09-28:
Thanks for replied @starbasessd 

This information can help?

https://learn.circuit.rocks/esp32-cam-with-rtsp-video-streaming

I tried to figured out, but, nothing

starbasessd, 2020-09-28:
Did you get the FTDI adapter, breadboard and associated stuff?
How far did you get in the setup instructions?
This camera is NOT for beginners, or even mid-level users, but for someone
with serious Arduino experience.
WAY beyond the level of support I can give you for free.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2020-09-28:
Yes, I bought this kit with all stuff including FTDI adapter.

https://www.ebay.co.uk/itm/1X-Esp32-Cam-Camera-Wifi-Bluetooth-Module-4M-Psram-Dual-Core-32-Bit-Cpu-De-W2E/224112725792?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

I get the setup and I can use the camera watching thru the ip address, opening google chrome webpage 

Basically, I followed up this steps from this video 

https://www.youtube.com/watch?v=ywCL9zDIrT0&t=2s

It was not difficult for me. I don't have any IT degree and zero experience on Arduino

I think it could be possible to watch the video streaming on MotionEyeOS with this camera, just I can not find the way. I mean because, this camera it doesn't have any encryption system that can be blocked to watch on MotioEyeOS and this ESP32-Cam it look like another IP camera like ONVIF camera and with ONVIF camera I can add to MotionEyeOS without any problem

Thanks

starbasessd, 2020-09-28:
According to the instructions, under:
Finally: The streaming functions.

The app has two video streaming options:

   - Streaming directly to a web browser
   - Streaming as an RTSP server (See Wikipedia
   <https://en.wikipedia.org/wiki/Real_Time_Streaming_Protocol> for an
   explanation of RTSP)

To select one of the two servers open main.h and enable the server you want
to run
you need to take out the # in front of

#define ENABLE_RTSPSERVER

to enable rtsp:.
Then you would select Network Cam in motionEyeOS when you add the camera,\
rtsp://<ipaddress>:8554/mjpeg/1
with whatever user and password you set up in the camera.

Really, any more and I need one to set up and test.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2020-09-28:
Thank you very much for replied.

Sorry how can I open main.h?

starbasessd, 2020-09-28:
Download the Arduino IDE software from here:
https://www.arduino.cc/en/main/software
Good luck.
I suggest getting a read wifi webcam:
https://www.amazon.com/s?k=wifi+webcam&i=electronics&rh=n%3A172282%2Cp_36%3A1253503011&s=price-asc-rank&dc&qid=1601332130&rnid=386442011&ref=sr_st_price-asc-rank

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2020-09-29:
Thank you very much @starbasessd

starbasessd, 2020-10-23:
Closing, no further information or updates supplied.

closed 2020-10-23 by starbasessd