motioneye-project/motioneyeos #2508: no video feed with a USB Video Capture Card

Zurd, 2020-09-09:
motionEyeOS Version: dev20200907
Board Model: Raspberry Pi 4B
Camera: Shrxy 800TVL CCD with an August VGB100 USB Video Capture Card
Network Connection: Ethernet

Cannot see a video feed, yet the camera is clearly loaded as we can see in dmesg:
[```
   96.062286] usb 1-1.3: new high-speed USB device number 4 using xhci_hcd
[   96.194682] usb 1-1.3: New USB device found, idVendor=1f4d, idProduct=0102, bcdDevice=40.02
[   96.194690] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   96.194697] usb 1-1.3: Product: Video Capture
[   96.194704] usb 1-1.3: Manufacturer: Geniatech Inc.
[   96.194710] usb 1-1.3: SerialNumber: 20090456
[   96.200603] cx231xx 1-1.3:1.1: New device Geniatech Inc. Video Capture @ 480 Mbps (1f4d:0102) with 6 interfaces
[   96.200659] cx231xx 1-1.3:1.1: Identified as Geniatech OTG102 (card=17)
[   96.201224] i2c i2c-12: Added multiplexed i2c bus 14
[   96.201357] i2c i2c-12: Added multiplexed i2c bus 15
[   96.376156] cx25840 11-0044: cx23102 A/V decoder found @ 0x88 (cx231xx #0-0)
[   96.392232] cx25840 11-0044: Direct firmware load for v4l-cx231xx-avcore-01.fw failed with error -2
[   96.392242] cx25840 11-0044: unable to open firmware v4l-cx231xx-avcore-01.fw
[   96.422241] cx231xx 1-1.3:1.1: v4l2 driver version 0.0.3
[   96.523358] cx231xx 1-1.3:1.1: Registered video device video0 [v4l2]
[   96.523441] cx231xx 1-1.3:1.1: Registered VBI device vbi0
[   96.523669] cx231xx 1-1.3:1.1: audio EndPoint Addr 0x83, Alternate settings: 3
[   96.523677] cx231xx 1-1.3:1.1: video EndPoint Addr 0x84, Alternate settings: 5
[   96.523686] cx231xx 1-1.3:1.1: VBI EndPoint Addr 0x85, Alternate settings: 2
[   96.523695] cx231xx 1-1.3:1.1: sliced CC EndPoint Addr 0x86, Alternate settings: 2
```
I choose Add Camera then Local V4L2 then Geniatech and it load the camera device as /dev/v4l/by-id/usb-Geniatech_Inc._Video_Capture_20090456-video-index0 but there is no feed. I even modified /data/etc/camera1.conf with videodevice /dev/vbi0, reboot but still nothing. It's just shows a gray square with a a camera in a Stop sign.

The camera works in Windows with the driver that it came and it also work in Linux Mint with the motion software.

starbasessd, 2020-09-09:
Its looking for firmware and not finding it.
[ 96.392242] cx25840 11-0044: unable to open firmware
v4l-cx231xx-avcore-01.fw
If you can find that version for ARMHF and you can scp it to the correct
location.
You mention it works in Linux Mint, was that on a Pi?
Does it work in RaspberryPiOS?
It could be that the particular firmware isn't available in RPiOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2020-09-09:
@Zurd does it work with Raspberry Pi OS (former Raspbian)?

Zurd, 2020-09-09:
Yes, it works with the latest Linux Mint but it is not a Pi version, just the regular Desktop version of linuxmint-20-mate-64bit.iso

I did not try it in Raspberry Pi OS because I don't have a micro-HDMI cable and by default there is no SSH access so I can't access. There must be a way to have SSH enabled by default, I will go look into it and report back.

starbasessd, 2020-09-09:
If you include your wpa_supplicant.conf file and a txt file called 'ssh' or
'ssh.txt' (with no content), in the /boot folder prior to first boot, ssh
is automatically enabled.
Image your microSDCard  <- I use Win32Disk Imager in Win10, but you can use
Etcher or Imager from RPiFoundation
copy the 2 files to /boot (shows as E: on mine, may vary on yours)
eject and insert in Pi
Boot normally
ssh to the pi (I use PuTTY, and use raspberrypi.local for the address for
RaspberryPiOS first boot, but you can use the IP address, too.)

This works for RaspberryPiOS, and motionEyeOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Zurd, 2020-09-09:
I found, too, how to ssh for a headless setup. It works. I'm into Raspberry Pi OS now and the camera is detected? I don't see any lines saying "unable to open firmware" like before.

[  110.060679] usb 1-1.4: new high-speed USB device number 4 using xhci_hcd
[  110.193236] usb 1-1.4: New USB device found, idVendor=1f4d, idProduct=0102, bcdDevice=40.02
[  110.193254] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  110.193271] usb 1-1.4: Product: Video Capture
[  110.193286] usb 1-1.4: Manufacturer: Geniatech Inc.
[  110.193300] usb 1-1.4: SerialNumber: 20090456
[  110.200555] cx231xx 1-1.4:1.1: New device Geniatech Inc. Video Capture @ 480 Mbps (1f4d:0102) with 6 interfaces
[  110.210661] cx231xx 1-1.4:1.1: Identified as Geniatech OTG102 (card=17)
[  110.211345] i2c i2c-12: Added multiplexed i2c bus 14
[  110.211497] i2c i2c-12: Added multiplexed i2c bus 15
[  110.389035] cx25840 11-0044: cx23102 A/V decoder found @ 0x88 (cx231xx #0-0)
[  112.315130] cx25840 11-0044: loaded v4l-cx231xx-avcore-01.fw firmware (16382 bytes)
[  112.346157] cx231xx 1-1.4:1.1: v4l2 driver version 0.0.3
[  112.454600] cx231xx 1-1.4:1.1: Registered video device video0 [v4l2]
[  112.454782] cx231xx 1-1.4:1.1: Registered VBI device vbi0
[  112.455269] cx231xx 1-1.4:1.1: audio EndPoint Addr 0x83, Alternate settings: 3
[  112.455286] cx231xx 1-1.4:1.1: video EndPoint Addr 0x84, Alternate settings: 5
[  112.455306] cx231xx 1-1.4:1.1: VBI EndPoint Addr 0x85, Alternate settings: 2
[  112.455323] cx231xx 1-1.4:1.1: sliced CC EndPoint Addr 0x86, Alternate settings: 2

I will install motion right now and see if it works, brb.

starbasessd, 2020-09-09:
If you get the error unable to load the firmware, try installing
RaspberryPiOS, then follow the instructions for installing motionEye on
Raspbian here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Then connect to the WebGUI at http://<yourIPAddress>:8765 and see if you
can add the capture card.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Zurd, 2020-09-09:
I confirm, it works in Raspberry Pi OS!

Seems like the good firmware is v4l-cx231xx-avcore-01.fw and it is missing in motionEyeOS.

Going to make a clean install of Raspberry Pi OS and install motionEyeOS on top of it.

starbasessd, 2020-09-09:
Not motioneyeOS. Follow the instructions at
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Install RaspberryPiOS, then install motion & motionEye on RaspberryPiOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Zurd, 2020-09-09:
Done. It is now installed and it is still not working! :(

motionEye is trying by default /dev/v4l/by-id/usb-Geniatech_Inc._Video_Capture_20090456-video-index0 which is a symlink to /dev/video0 which is perfect but I still don't see anything.

starbasessd, 2020-09-09:
In the WebGUI, can you select add new camera, type: V4L2, and look for the
video0 or video10?
Also, enable VNC, in raspi-config,
Then from another computer, connect via vnc-viewer, and see if raspivid or
raspicam work with the capture?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-09-09:
BTW, I feel your pain. I have an USB AverMedia tv/video capture device that
works on everything but my Pis. No driver for ARM processors. Still looking
for a good one.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Zurd, 2020-09-09:
> BTW, I feel your pain. I have an USB AverMedia tv/video capture device that works on everything but my Pis. No driver for ARM processors. Still looking for a good one.

Not the case here, like I said, it is working in Raspberry Pi OS with motion. The drivers are there and loaded.

starbasessd, 2020-09-09:
But not working in motionEye on that setup?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Zurd, 2020-09-09:
Exactly, not working with motionEye installed on top of Raspberry Pi OS but I managed to make it work with only Raspberry Pi OS and motion.

Zurd, 2020-09-09:
I managed to make it fully work on the same OS as motionEye is installed with motion only but whatever I try, it doesn't work in motionEye.

There is no error in /var/log/motion.log

Just a note, it only works in 320 x 240 in motion, else motion keeps restarting and I have this error: motion_watchdog: Thread 1 - Watchdog timeout. Trying to do a graceful restart

starbasessd, 2020-10-23:
Status?

Zurd, 2020-10-23:
Same status, nothing changed. I didn't try a new version of motionEye if that is what you are asking.

starbasessd, 2020-10-23:
What is your config file for motion and the card?

Zurd, 2020-10-23:
No idea, everything is formatted clean. If you ask me for the specific path location of the file, next time I play with it, I will be able to tell you.

Zurd, 2020-12-13:
Just tried motioneyeos-raspberrypi4-dev20201026.img and same problem, camera is not detected.

dmesg reports:
```
[  319.246406] usb 1-1.1: new high-speed USB device number 4 using xhci_hcd
[  319.389249] usb 1-1.1: New USB device found, idVendor=1f4d, idProduct=0102, bcdDevice=40.02
[  319.389270] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  319.389287] usb 1-1.1: Product: Video Capture
[  319.389303] usb 1-1.1: Manufacturer: Geniatech Inc.
[  319.389319] usb 1-1.1: SerialNumber: 20090456
[  319.399595] cx231xx 1-1.1:1.1: New device Geniatech Inc. Video Capture @ 480 Mbps (1f4d:0102) with 6 interfaces
[  319.399947] cx231xx 1-1.1:1.1: Identified as Geniatech OTG102 (card=17)
[  319.402070] i2c i2c-12: Added multiplexed i2c bus 14
[  319.402646] i2c i2c-12: Added multiplexed i2c bus 15
[  319.600134] cx25840 11-0044: cx23102 A/V decoder found @ 0x88 (cx231xx #0-0)
**[  319.617919] cx25840 11-0044: Direct firmware load for v4l-cx231xx-avcore-01.fw failed with error -2
[  319.617937] cx25840 11-0044: unable to open firmware v4l-cx231xx-avcore-01.fw**
[  319.650590] cx231xx 1-1.1:1.1: v4l2 driver version 0.0.3
[  319.755721] cx231xx 1-1.1:1.1: Registered video device video0 [v4l2]
[  319.756048] cx231xx 1-1.1:1.1: Registered VBI device vbi0
[  319.756987] cx231xx 1-1.1:1.1: audio EndPoint Addr 0x83, Alternate settings: 3
[  319.757004] cx231xx 1-1.1:1.1: video EndPoint Addr 0x84, Alternate settings: 5
[  319.757022] cx231xx 1-1.1:1.1: VBI EndPoint Addr 0x85, Alternate settings: 2
[  319.757038] cx231xx 1-1.1:1.1: sliced CC EndPoint Addr 0x86, Alternate settings: 2
```

And as usual, still working with Raspberry Pi OS (32-bit) released on 2020-12-02 with only motion 4.3.2:
```

[  222.295600] usb 1-1.2: new high-speed USB device number 3 using xhci_hcd
[  222.428441] usb 1-1.2: New USB device found, idVendor=1f4d, idProduct=0102, bcdDevice=40.02
[  222.428460] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  222.428477] usb 1-1.2: Product: Video Capture
[  222.428492] usb 1-1.2: Manufacturer: Geniatech Inc.
[  222.428507] usb 1-1.2: SerialNumber: 20090456
[  222.532291] cx231xx 1-1.2:1.1: New device Geniatech Inc. Video Capture @ 480 Mbps (1f4d:0102) with 6 interfaces
[  222.532422] cx231xx 1-1.2:1.1: Identified as Geniatech OTG102 (card=17)
[  222.533387] i2c i2c-12: Added multiplexed i2c bus 14
[  222.533592] i2c i2c-12: Added multiplexed i2c bus 15
[  222.725181] cx25840 11-0044: cx23102 A/V decoder found @ 0x88 (cx231xx #0-0)
**[  224.714446] cx25840 11-0044: loaded v4l-cx231xx-avcore-01.fw firmware (16382 bytes)**
[  224.748227] cx231xx 1-1.2:1.1: v4l2 driver version 0.0.3
[  224.854699] cx231xx 1-1.2:1.1: Registered video device video0 [v4l2]
[  224.855205] cx231xx 1-1.2:1.1: Registered VBI device vbi0
[  224.855223] cx231xx 1-1.2:1.1: video EndPoint Addr 0x84, Alternate settings: 5
[  224.855241] cx231xx 1-1.2:1.1: VBI EndPoint Addr 0x85, Alternate settings: 2
[  224.855258] cx231xx 1-1.2:1.1: sliced CC EndPoint Addr 0x86, Alternate settings: 2
[  224.856157] usbcore: registered new interface driver cx231xx
[  224.875029] cx231xx 1-1.2:1.1: audio EndPoint Addr 0x83, Alternate settings: 3
[  224.875046] cx231xx 1-1.2:1.1: Cx231xx Audio Extension initialized
```

Would have been fun with motionEye, easier to configure but I will just stick with motion, I just finished configuring the whole thing so not going to bother anymore with motionEye or motionEyeOS.