motioneye-project/motioneyeos #2510: Camera stop working after couple of minutes

Mike1236, 2020-09-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606

### Board Model

I am using the following board/model: I have Raspberry Pi 4 2GB as a hub and Raspberry Pi Zero W as the camera

### Camera

I am using the following type of camera: Fast Network Camera

My camera model is: Raspberry Pi NoIR V2

### Network Connection

My motionEyeOS unit is connected to the network via: Hub is ethernet and Camera on Pi Zero W is wireless

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

No peripherals at the moment

### Log Files

I consider the following log files relevant to this issue:

[boot.log](https://github.com/ccrisan/motioneyeos/files/5203409/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5203411/motioneye.log)

I can attach more logs if necessary

starbasessd, 2020-09-10:
What are your symptoms?
Does the Camera stop working on the PiZero, or do you lose connectivity to
the camera from the hub?
If you lose connectivity, can you still get to the camera from another
browser?
20200606 has multiple known issues with Pi4. Can I recommend upgrading to
20200907 on both?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
So of course right now it's been running for 15 minutes without any issues. However, before when it would stop working around 5 minutes. I have the PiZero connected via HDMI so I can see logs. When it would stop working it would spit a whole page of errors? I assume they are errors. The hub works fine but it would show the camera has disconnected. I cannot access the camera directly with any browser. I see that 20200907 is pre-release, but I can certainly try it if needed. I do see that the CPU on the PiZero is being taxed heavily.
When I run uptime it spits out:

13:03:50 up 17 min,  load average: 2.00, 2.06, 1.50

Should I try to overclock the PiZero?

starbasessd, 2020-09-10:
No, overclocking puts even more stress on the PiZero.
On the PiZero:
Can you give the messages log?
Can you give the results for
vcgencmd measure_temerature
vcgencmd get_throttled
Maybe a phone camera screenshot of the error messages on the screen?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
So it crashed again but I didn't have the HDMI connected to the monitor so I wasn't able to see the messages. However I rebooted it by unplugging and replugging the power, it shows some messages similar to what it shows when it crashes. 

[image]
[image]
[messages.log](https://github.com/ccrisan/motioneyeos/files/5203721/messages.log)

starbasessd, 2020-09-10:
Oops, sorry about that
vcgencmd measure_temp

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
[image]

starbasessd, 2020-09-10:
The messages logs also indicate it is in the mid 60sC, kinda warm, and may
be warm enough to cause problems. Is it in a case? Any chance to increase
airflow?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
It's in the official raspberry pi zero w case. I'm not sure how I can increase airflow with this case.

starbasessd, 2020-09-10:
It was a thought.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
I'll try to upgrade it to 20200907. Do I just reflash the microSD with the newer image?

starbasessd, 2020-09-10:
You can do a backup/restore, but yes, re-image is fastest.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
So I have it as 20200907 now. However I cannot add it to my hub.

[image]

Mike1236, 2020-09-10:
I'm going to try 20200203

starbasessd, 2020-09-10:
Try with port 8081 or as simple mjpeg camera.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-09-10:
Also did you turn on fast network camera?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
Seems like I forgot to add the port 8081. However I found this out after I downgraded to 20200203.

Mike1236, 2020-09-10:
So it just crashed again, wasn't able to get the screen messages :(

Mike1236, 2020-09-10:
I'm going to try 20190911 now

starbasessd, 2020-09-10:
I just built one with RpiZeroW, 16GB MicroSDCard, PiCam v1.3, Official Case
(with camera mounted in closed lid), 20200907 set to Fast Network cam
Getting temp of 41.2C
No issues with crashing
I'm beginning to think your Pi may be the issue.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
Hm okay, the PiZeroW is brand new, I just bought it. I guess I should go to Microcenter where I bought it and exchange it.

Mike1236, 2020-09-10:
I don't think there is enough room for the heatsink if I have the camera module installed too. I'm going to try the PiOS and see what the temps are then before exchanging.

starbasessd, 2020-09-10:
Let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
So in order to power this PiZero, I have to plug then unplug, then plug the power for it to power on. I assume this isn't normal? It showed 45.1C for temps, but I had it open with no case. So I'm trying with case now.

Mike1236, 2020-09-10:
Looks like the temps are the same with the case on.

Mike1236, 2020-09-10:
I'm going to try 20200907 again with just Fast Network Cam.

starbasessd, 2020-09-10:
I'll bet the temps are back up.
It takes a bit (up to a couple minutes if using a very large SDCard) for it
to set itself up, and would reboot automatically. for RPiOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-09-10:
> So I got it running 20200907 with just Fast Network Cam. For some reason
> this dashboard page is grayed out and I cannot access it. Temps is also
> sitting at 60C. I'm thinking this may be a faulty RPiZeroW now....
>
> [image: dashboard]
> <https://user-images.githubusercontent.com/60204760/92796132-ec0e2400-f37e-11ea-9844-f481e52396c3.png>

Not logged in as admin.
Most is disabled, anyway, you do all the hard work on the Hub Pi.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
It is still grayed out even though I logged in as admin. Does red border on the camera mean that it's recording?

starbasessd, 2020-09-10:
Something is wrong. Try rebooting.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
Hmm I was able to edit settings at both the hub and camera level before, now it only works on the hub. I'm logged in as admin. Temps are still sitting at 60C. However I noticed now that the monitor flickers on and off when the HDMI cord is now level. The site where plugged into the PiZero.

starbasessd, 2020-09-10:
Yep, something is definitely wrong.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
So something creepy just happened on the monitor. It flashes 3 times with this image.

[image]

starbasessd, 2020-09-10:
I vote for getting a replacement.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
Yeah, I'm going to head to Microcenter in 45 minutes to get an exchange. Hopefully it works with a new one.

starbasessd, 2020-09-10:
I just swapped the RPiZeroW with 3 others I have sitting here, and all are
within a few degrees of each other. (40-45C) all running the same camera
and SDCard and power supply.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
Yeah so my hub which I'm using Raspberry Pi 4 2GB, it has heatsinks and a small fan. I'm using this case: https://www.amazon.com/gp/product/B07VX3HQGJ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

It's sitting idle at 37C. So I think you're right, something is definitely wrong with the PiZeroW I bought haha

starbasessd, 2020-09-10:
BTW, for that one, I'd seriously recommend 20200907 for all the bug fixes
with the earlier versions of the Pi4's. I have a 4GB and 8GB, and it's much
better.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-10:
I see. I'll reimage all the SD cards when I get back with the new PiZero. I have nothing to lose since I just start this project today :D

Mike1236, 2020-09-11:
So I went to Microcenter and exchanged for another PiZeroW.

Reimaged RPi4 (hub) and the new RPiZeroW with 20200907 respectively. The RPiZeroW is running hotter than before averaging 65C. However I set the settings to record 1280x720@10fps with 25% image quality and Fast Network Camera ON.

It's been running for 20 minutes straight now and it hasn't crashed since. It seems stable. I'm still kind of worried about the temps, however I do have to say the room temperature is 78F right now.

Mike1236, 2020-09-11:
Now I have a different problem. I'm not sure what is causing it. The is a green bar on the top of the video recordings. I've changed out the CSI cables.

[image]

Not sure what is causing this issue now....

starbasessd, 2020-09-11:
Is that on the PIZero, or the hub?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-11:
Hub has no camera attached to it. I'm viewing the PiZero recordings on the hub side.

starbasessd, 2020-09-11:
are you seeing it on the hub only or on both? If you look at the
http://PiZero:8081 with a browser, is it there? Or only when you look at
the hub image?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-11:
Stream has no green bar on the PiZero and the hub. I only see it when viewing the recorded videos. I also downloaded the recording and it shows there. So it must be something wrong with the recording itself and not the camera or stream? I've also switched out the PiCam NoIR to a regular PiCam V2 thinking that may be the problem but it wasn't, still green bar in recordings.

starbasessd, 2020-09-11:
Wow. Interesting. I'm going to flag @ccrisan and @jasaw about this as I
have no idea.
A couple things to try:
Lower resolution
Lower frame rate
Set Movie Passthrough
or set different format

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-09-11:
I don't know what happened just now that fixed it? I lowered the resolution from 1280x720 to 1024x768. Frame rate stayed the same at 10fps. The green bar disappeared in the recordings. So I changed it back to 1280x720@10fps and the green bar isn't there anymore. So maybe it was a hiccup? I'm extremely appreciative for the help btw! I think this new PiZero doesn't seem like it's going to crash even though the temps is sitting at 60C average right now.

Mike1236, 2020-09-11:
I found the culprit. I changed the movie recordings from H.264/OMX to H.264. That fixed the problem!

starbasessd, 2020-09-11:
Good deal, glad to help.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Resolved

closed 2020-10-23 by starbasessd