motioneye-project/motioneyeos #2511: (question) is there any dynamic dns solution incorporated to the OS? If not, Can I use DuckDNS in it?

Lion4K, 2020-09-10:


starbasessd, 2020-09-10:
If you can use their client on any other PC on the network, I would do it
that way, but that's my opinion, only. AFAIK, there isn't any DDNS enabled
on motionEyeOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Lion4K, 2020-09-10:
I see.
Well the install instructions tells me to look for these features.

"if your linux install is running a crontab, then you can use a cron job to keep updated"

I believe I can run it from motioneyeOS itself no?

starbasessd, 2020-09-10:
One issue with motiojnEyeOS: you can't add other packages with apt or dpkg,
as neither are installed. If it's a straight python script that supports
python2.7, and no extra modules, yes.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Lion4K, 2020-09-11:
Well, from what I saw I need cron and curl.
Are these available?

Lion4K, 2020-09-11:
So I just tested and I was able to get it working, so if you wanna tell others can use duck dns, you can tell it works.

starbasessd, 2020-09-11:
So share, what did you need to do, so it can be added to the Wiki.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Lion4K, 2020-09-12:
I actually just followed the instructions for Linux Cron.
I'm not gonna post the instructions here because I understand that the instructions might change over time and or motioneyeOS might get striped of cron or curl breaking this particular instructions. And since google (sometimes) indexes answers here, it could in the future misguide someone.
But as you can check, on their website (duck dns) they have instructions for several OS'es and Systems [image] I haven't tested but I'm sure more than one of those would work on motioneyeOS.

starbasessd, 2020-09-12:
Looking through links, I did find this reference in the Wiki, Tweaks:
https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support
For example, the line updates the IP address on freedns.afraid.org dynamic
DNS service, every 5 minutes:

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Resolved

closed 2020-10-23 by starbasessd