motioneye-project/motioneyeos #2512: (question) if I add an usb microphone or a camera with microphone can I record the audio from whenever it records ?

Lion4K, 2020-09-12:


starbasessd, 2020-09-12:
Not currently. In the issues, several have tries script mods, there has
been some discussion, but IIRC it has never been able to be confirmed
working across the board. I believe its also an active request in the
motion  groups, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Lion4K, 2020-09-12:
I see.

Thanks for the clarification

originalknibbler, 2020-10-26:
Maybe it possible to add audio with an additional selfmade script like here: https://blog.lgohlke.de/2014/02/09/babyphone-mit-dem-raspberry-pi.html so that you have the sound on an other tab and an other port

starbasessd, 2020-10-26:
Not with motionEyeOS. You might try with RaspberryPiOS and motion/motioneye
install and those instructions.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-26:
Here are the instructions for RaspberryPiOS & motion/motionEye install:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?