motioneye-project/motioneyeos #2513: Streaming wont work after a few minutes

dabos1, 2020-09-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
yes
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
yes
### motionEyeOS Version

I am running motionEyeOS version: lates Version

### Board Model

I am using the following board/model: Ubuntu 18 on an VM 

### Camera

I am using the following type of camera: Network Camera(Esp32 Cam)

My camera model is: Esp32 cam

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
Hello!

I just install Motioneye on my Ubuntu 18 machine. And i added a IP network camera and it works for a few minutes. But than after a few minutes the stream will stop. Furthermore the motion detection works just fine the whole time.

starbasessd, 2020-09-13:
Which motionEyeOS? Specifically? 20190911, 20200203, 20200606, 20200907?
Or are your running motion/motionEye install? If this, did you run the
setup instructions from here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu
When you say the stream stops, what is the indication?
From another computer or phone web browser (on the same network) can you
connect to http://ip_of_vm:8081? Can you see anything?
Which VM environment are you using? Virtualbox, VMWare, eSXI, other?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dabos1, 2020-09-13:
> http://ip_of_vm:8081

Yes I followed the instruction. The stream stops when i don't see the stream in the motioneye web-interface. When i connect to http://ip_of_vm:8081 I see the stream for the first few minutes and than it will stop. I running Ubuntu in an LXC environment

starbasessd, 2020-09-13:
Are you using Edge for your web browser?
Can you answer the rest of the questions, please?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dabos1, 2020-09-13:
> Are you using Edge for your web browser? Can you answer the rest of the questions, please?
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

I use the Brave browser. And it seems that the stream stops when an motion is detected

starbasessd, 2020-09-13:
I am not familiar with LXC, I will try to build an environment.
I use VBox mostly but am familiar with eSXI and VMWare.
Is your LXC host also Ubuntu, and which version?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dabos1, 2020-09-13:
> I am not familiar with LXC, I will try to build an environment. I use VBox mostly but am familiar with eSXI and VMWare. Is your LXC host also Ubuntu, and which version?
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

As host I use Proxmox

starbasessd, 2020-09-13:
Brave on a PC, MAC, Ubuntu, other? Is it an apt install?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dabos1, 2020-09-13:
> Brave on a PC, MAC, Ubuntu, other? Is it an apt install?
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

On Ubuntu 20 and it was installed with apt

starbasessd, 2020-09-13:
I'm sorry, but I don't have any spare iron to install a whole new server
environment on top of all the other stuff, so I am going to have to back
off and see if anyone else can help.
I know it works in VBox and VMWare/eSXI VMs. Some have had it working in
Docker environments. Works on many flavors of Linux, and on several (14 at
last build) of SBCs.
I can't recreate your specific environment to help, though.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dabos1, 2020-09-13:
> I'm sorry, but I don't have any spare iron to install a whole new server environment on top of all the other stuff, so I am going to have to back off and see if anyone else can help. I know it works in VBox and VMWare/eSXI VMs. Some have had it working in Docker environments. Works on many flavors of Linux, and on several (14 at last build) of SBCs. I can't recreate your specific environment to help, though.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

I thank the issue is not related to my environment. Because the stream will stop working when an motion is detected

starbasessd, 2020-09-13:
The Proxmox specs say only the Big 4 browsers are supported for their
interface. Can you please try Safari, Chrome, Edge, or FireFox and see if
the same issue occurs?
There are issues with the latest Edge, but FF and Chrome are supported in
Ubuntu 18 & 20 if viewing from the VM

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dabos1, 2020-09-13:
I get the same issue when I use Firefox, it doesn't change anything. Every time a motion is detected the streaming will stop

starbasessd, 2020-09-13:
OK, sorry, but I cannot re-create the environment and can't get to root
cause in your situation.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dabos1, 2020-09-13:
> OK, sorry, but I cannot re-create the environment and can't get to root cause in your situation.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

OK, Than I will try another software

starbasessd, 2020-10-23:
Closed, no further information.

closed 2020-10-23 by starbasessd