motioneye-project/motioneyeos #2514: Lost control over the web interface

ebolisa, 2020-09-14:
Hi,

I had to reinstall motion and I lost control over the web interface. Where should I start to look?
TIA

Port 8080 (shown in motion.conf) is streaming OK.
Port 8081 (where the motionEye web should come up, I get what is shown in the img. Port 80 is used up.
Raspberry 3 B+ with Buster. No SD but a HDD.
 
[image]

starbasessd, 2020-09-14:
Probably 8765

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ebolisa, 2020-09-14:
Thanks for the quick response. Port 8765 is not responding.

starbasessd, 2020-09-14:
Did you re-install motionEyem too?
There is another motion port fi enabled 7999

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ebolisa, 2020-09-14:
No, I did not re-install MotionEye.

starbasessd, 2020-09-14:
You may want to go through the full steps for motionEye Install here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ebolisa, 2020-09-14:
Will do. Thank you.

ebolisa, 2020-09-14:
OK, regained control of the web UI. Thank you much!! Even though there were no errors during the process, I get the following when checking the service status. Is it some to worry about?
```

 sudo systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-09-14 17:30:14 CEST; 3min 8s ago
 Main PID: 11587 (meyectl)
    Tasks: 8 (limit: 1939)
   CGroup: /system.slice/motioneye.service
           ├─11587 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
           ├─11637 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
           └─13169 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5

Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "port"
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "threshold_maximum"
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_passthrough"
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "mmalcam_name"
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [NTC] [ALL] motion_startup: Logging to syslog
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
```
These lines are in red.

```

Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "port"
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "threshold_maximum"
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_passthrough"
Sep 14 17:32:53 rpicasa motion[13169]: [6874760:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "mmalcam_name"

```

starbasessd, 2020-09-14:
Did you manually modify :

/etc/motioneye/motion.conf

in mine on an Ubuntu Buster, I don't have any of those options there. They
are in the individual camera-x.conf files in /etc/motioneye/ folder.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ebolisa, 2020-09-14:
No. It's a copy using the cmd `cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf`. I just changed the port since I cannot use 80.

ebolisa, 2020-09-14:
Just find out that it's a question of time. When I check on the service again few minutes later, every thing is OK. Don't ask me why.

s```
udo systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset
   Active: active (running) since Mon 2020-09-14 18:00:27 CEST; 14min ago
 Main PID: 11359 (meyectl)
    Tasks: 16 (limit: 1939)
   CGroup: /system.slice/motioneye.service
           ├─11359 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/mo
           ├─11387 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5
           └─11415 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/mo

Sep 14 18:00:29 rpicasa motion[11387]: [23357064:motion] [NTC] [ALL] motion_star
Sep 14 18:00:29 rpicasa motion[11387]: [23357064:motion] [NTC] [ALL] motion_star
Sep 14 18:00:29 rpicasa motion[11387]: [23357064:motion] [NTC] [ALL] motion_star
Sep 14 18:00:29 rpicasa motion[11387]: [23357064:motion] [NTC] [ALL] motion_star
Sep 14 18:00:31 rpicasa meyectl[11359]:     INFO: cleanup started
Sep 14 18:00:31 rpicasa meyectl[11359]:     INFO: wsswitch started
Sep 14 18:00:31 rpicasa meyectl[11359]:     INFO: tasks started
Sep 14 18:00:31 rpicasa meyectl[11359]:     INFO: mjpg client garbage collector
Sep 14 18:00:31 rpicasa meyectl[11359]:     INFO: server started
Sep 14 18:00:55 rpicasa motion[11387]: [1:ml1:Camera1] [WRN] [ENC] ffmpeg_set_co](url)
```

starbasessd, 2020-09-14:
The /etc/motioneye/motion.conf file is the one that is indicating it was
changed/modified, and should not have been by those errors.
.Here's the content of my /etc/motioneye/motion.conf file:
-----
cat /etc/motioneye/motion.conf
# @enabled on
# @show_advanced on
# @normal_password
# @admin_username admin
# @admin_password
# @normal_username user

thread thread-1.conf
webcontrol_port 7999
setup_mode off
webcontrol_parms 2
webcontrol_localhost off
webcontrol_interface 1

camera camera-2.conf
camera camera-3.conf
camera camera-4.conf
camera camera-1.conf
camera camera-5.conf
-----
By default /etc/motioeye/motioneye.conf uses port 8765 on a non-motionEyeOS
setup, so it shouldn't have been changed, either.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ebolisa, 2020-09-14:
Got it! Thanks again!!

closed 2020-09-14 by ebolisa