motioneye-project/motioneyeos #2516: Fullscreen does not work properly unless all camera feeds are active

fedmartinez, 2020-09-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

motionEyeOS 20200606

### Board Model

Raspberry Pi 4 Model B

### Camera

1. Simple MJPEG Camera (Raspberry Pi Zero W + Raspberry Pi Camera Board v2 - 8 Megapixels
2. Simple MJPEG Camera (Raspberry Pi Zero W + Raspberry Pi NoIR Camera Board v2 - 8 Megapixels 

### Network Connection

Ethernet

### Peripherals

N/A

### Screenshot

[image]

### Description

I am utilizing a RPI4B as a central server which connects to two Simple MJPEG cameras via my local network, Anytime I pull up the central server and attempt to view one of my cameras in full-screen mode (through the MotionEye OS UI itself), the camera feed will take up a minuscule amount of the window. This is only the case when the second camera is shut off; when both cameras are turned on and active, this is NOT an issue and full-screen functionality works as expected.

starbasessd, 2020-09-15:
THere are a number of issues with 20200606. Can you downgrade to 20200203
or 20190911, or upgrade to dev20200907?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Status?