motioneye-project/motioneyeos #2517: Toggle GPIO during motion detection event.

Lion4K, 2020-09-16:
I was thinking on adding like a led light strip to the front of the camera field of view and having it toggled on to kinda scare whoever gets in it's FOV.
I was thinking on hooking the led strip to the GPIO either directly or with some relays if the power supply isn't sufficient.
Is there a way to set this up?
I saw something about custom command on the motion detection settings.

starbasessd, 2020-09-16:
Theoretically, under Motion Detection, Run Command could trigger a script
that could turn on a GPIO pin, wait however long, and turn it back off. I
have not tried it, myself.
There is also a Run Command under File Storage that runs after the media
file is created Which is probably too late for your purpose.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Lion4K, 2020-09-16:
I see but either way I can't command the GPIO directly, I need to set up a script that will do that, right?

starbasessd, 2020-09-16:
Again, theoretically, according to the tool tips for the command lines you
could do something like the following as a single line:
cmd to turn on GPIO pin; wait 10;  cmd to turn off GPIO pin
to turn on the pin, leave on for 10 seconds and turn it back off, but I
don't know how that would affect the recording.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?