motioneye-project/motioneyeos #2518: Raspberry Pi 4 8GB - cannot install - “This board requires newer software”

apsvett, 2020-09-17:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

Raspberry Pi 4 8Gb.

I am using the following board/model: Raspberry Pi 4 8Gb.

### Camera

I am using the following type of camera:  Network Camera

My camera model is: None yet

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

[image]

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

motionEyeOS is not possible to install at the 8Gb-version of Pi4.  4Gb-board works.

Error message at boot: “This board requires newer software”

starbasessd, 2020-09-17:
2 solutions:
Download and install dev20200907 from here:
https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev
or add the attached .zipped files to /boot (like adding wpa_supplicant.conf and ssh)
The zipped files do not fix an issue with USB attached cameras, though.

[RPi4-Boot-Files.zip](https://github.com/ccrisan/motioneyeos/files/5239310/RPi4-Boot-Files.zip)

apsvett, 2020-09-17:
There, that does the trick. Thanks a lot, starbasessd !

closed 2020-09-17 by apsvett

tjerkkamstra, 2021-05-05:
The solution starbasessd offered works like a charm. But now the questions arrises here: I have two similar Pi 4's (one's a 8gb the other a 2 gb version): why does it work without the fix on the 8gb board and does the 2 gb board need this fix?

starbasessd, 2021-05-05:
The current solution is to bring the bootloader up to current stable,
following the instructions here:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md
(several options listed) and run either motioneyeos version dev20201026
(prefered) or 20200606 (a few issues affecting the Pi4s)
The Pi4s (all versions) seldom ship with the latest version of the boot
loaders. If you ran RaspberryPiOS, and updated/full upgraded that is one of
the ways to get the latest boot loader...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tjerkkamstra, 2021-05-05:
I should add: tried to get it working. It indeed booted with those new files, but it failed to recognise the camera.
What did work is downloading the dev-version. That did not need those two files: it booted directly after burning the image onto the SD card.