motioneye-project/motioneyeos #2519: dropbox upload issues

jarmodb, 2020-09-18:
### motionEyeOS Version

I am running motionEyeOS version:20200606

### Board Model

I am using the following board/model: 3 times rpi zeroW as cams, 1 rpi 4, 4GB as main hub
### Camera

I am using the following type of camera: fast network camera

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: pi zero's via wifi, rpi4 via ethernet
### Peripherals

hi all, i have some problems with dropbox uploads last months. It was running good for about half an year, and since a few months ago my files sometimes stop uploading to dropbox, i have to delete motioneye oploader from my connected dropbox apps, and obtain a new authorization key in motion eye os... is there a fix for this problem?

starbasessd, 2020-10-01:
Please attach motion, motioneye, and messages logs from the RPi4

starbasessd, 2020-10-23:
Closing. Re-open or open a new issue when you can attach the motion, motioneye, and messages logs from the RPi4

closed 2020-10-23 by starbasessd