motioneye-project/motioneyeos #2520: Motion Eye Crashes!!!

STUKguy, 2020-09-19:
Hi,

Device: Raspberry PI4 2GB
motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

Not sure why, but for some reason my motion eye keeps crashing, it will work fine for a couple of hours and then just crashs, thought it was a faulty SD card so put a new one in with a freash build of MotionEye on it but still the same, also i have it on a 7 day record and delete, the delete part does not work either i have to go in and manually delete the files

Here is my log if anyone can help?
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5249906/motioneye.log)

Thanks,

starbasessd, 2020-09-19:
There are a number of known issues with 0606. My recommendation is to run
either 20190911, 20200203, or dev20200907, all of which are much more
stable for me, especially on RPi4-4GB & RPi4-8GB, and the last has major
updates to kernel, and motion/motionEye.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-19:
@starbasessd thanks for your reply i will try dev20200907 now and if need be try the others after, with these be "DEV" does that mean i will need to update the more regulary that the other type of release?

STUKguy, 2020-09-19:
@starbasessd With dev20200907 not sure what going on but the cameras keeo cutting in and out and i have no access to the the menu
[image]
 it just stays like this

starbasessd, 2020-09-19:
I don't know, bad image write, maybe?
 It's working on one of my RPi4-4GB and on my RPi4-8GB. No issues, clean
install, from download here:
https://github.com/ccrisan/motioneyeos/releases

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-19:
@starbasessd so i redid the image still the same issue, downloaded it again still the same so went back to 20200203 and that seems to work fine (ish) camera's cut out now and again. i have set it up and see how it goes over the next few days.

Question, would 2 usb camera's and 1 RTSP feed be too much for a Raspberry PI 4 2GB to handle?

starbasessd, 2020-09-19:
No. I do (on average) 5-6 cameras, 1 or 2 USB (different brand/models) 3 IP
network cameras (D-Link) 1 RTSP (FosCam) and I have a PiCam that I put on a
Zero, 2, 3 or 4 for additional testing.
(My master is a VM of Ubuntu 20.04 with the 3 D-Links, the RTSP, and a USB
on it)
I can usually re-create most people's environments. No issues on my master
VM or the testing boxes.
I have issues where it won't mount one or the other camera if I put 2 USB
cameras with the same make/model on the same motionEyeOS though.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-19:
@starbasessd that's good to know thank you, so can Motion eye be ran in a VM? Tried running Motion eye in a docker (OMV) but could not figure out how mount the USB camera.

starbasessd, 2020-09-19:
I'm currently running an Ubuntu 20.04 host on Ryzen 1700 (8 cores) with
32GB RAM & 10TB storage.
I'm running VirtualBox 6.1.14 with a number of virtual machines
My Master motionEye is running Ubuntu 20.04 as guest with pass through USB
and handles it just fine. along with shared folders to the host 10TB.
I started it back with Ubuntu 18.04 both host and guest.
Ask me almost anything, I'll try to answer or tell you I am clueless.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-19:
@starbasessd so theoretically if I was to run a Windows machine (half the specs you have) with a VM of Ubuntu running Motion eye, would that work?

starbasessd, 2020-09-19:
I would think so, and nothing to lose but an hour or so of your time to try.
Even my Windows 10 laptop with an AMD FX8800 can run the VM, but not the
USB pass through due to driver issues. (HP laptop using weird USB that
Windows knows but VBox doesn't like)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-19:
@starbasessd, Thank you so much for this. I'm slowly starting to lose my patience with Raspberry pi, but for the moment I'm stuck to using them currently I am having issues trying to install Next cloud on docker (OMV) but if I can get hold of a powered USB hub to use on my PI I might give trying to get Motion eye on docker (OMV) to work

starbasessd, 2020-09-19:
Just for grins and giggles, would you check a couple of things for me,
please?
Run these 2 commands on motionEyeOS or RaspberryPiOS?
vcgencmd measure_temp
vcgencmd get_throttled
and let me know the results, please?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-19:
Sure 
[root@CCTV ~]# vcgencmd measure_temp
temp=59.0'C
[root@CCTV ~]# vcgencmd get_throttled
throttled=0x0

starbasessd, 2020-09-19:
Temp is a bit high, do you have any way to ventilate, or maybe put
heatsinks in, or both? I notice that my 3B+ and 2s have heat issues over
55C, and my Zeros start having issues over 50C. My 4-4GB and 4-8GB have
active cooling (fans and heasinks) and seldom get above 40C.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-09-19:
On the other hand, looks like your power supply is keeping up, if you have
a USB camera connected too. (throttled=0x0 is good, no low power incidents
since turn on.)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-19:
@starbasessd so some back story behind my Motion PI set up I brought a X820 expansion board an case for my PI under the assumption that I could boot Motion eye from an SSD rather than a SD card, then I did more research and found it is not possible but kept the X820 but it came with a fan which for the last 3 weeks I've been saying to my self I need to move the fan due to the heat but just never get around to it. I'm using the original PI 4 plug (5V 3AMPs) just modded slightly to work with the X820

[image]

Bare In mind the X820 is for the PI 3 but with a slight change works with the PI4

starbasessd, 2020-09-19:
Why can't you get it to boot to the SSD, out of curiosity? My 3Bs and 3B+s
all either USB or NetBoot just fine. My 2s and Zeros run off of USB drives
(only /boot on a SDCard).

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-19:
The commands I found only were only for booting Rapbian from SSD and when I tried them on Motion eye os it refused to accept them, and then I found a form post from the creator of Motion eye saying he was interested in adding this functionality to Motion eye os. I'm assuming the command to do this with a PI3 would be different to the PI4

starbasessd, 2020-09-19:
Oh, missed the part about booting motionEyeOS to SSD. Doesn't work for USB,
either. THere's too much hard coded to the SDCard partitions, instead of
like Raspbian, that you can edit or it auto-discovers where the partitions
are. Frustrating. Sorry.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-19:
@starbasessd no worries, Thank you again for your help.

starbasessd, 2020-09-19:
If you choose Ubuntu for your guest use these:
https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu
Anything else, look here:
https://github.com/ccrisan/motioneye/wiki/Installation

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-20:
@starbasessd so I finnaly got around to dismounting the fan from the case and made a small standing

[image]

This is temp I got
[root@CCTV ~]# vcgencmd measure_temp
temp=65.0'C

Flipped the fan over (as I thought I might have the fan upside down) got this temp
[root@CCTV ~]# vcgencmd measure_temp
temp=68.0'C

Replaced the fan with the one from my PI NAS which is roughly the same size, I got this temp
[root@CCTV ~]# vcgencmd measure_temp
temp=53.0'C

Plugged the motion eye case 
pi@LA-NAS:~ $ vcgencmd measure_temp

temp=39.0'C

pi@LA-NAS:~ $ vcgencmd measure_temp

temp=40.0'C

starbasessd, 2020-09-20:
Those last 2 are really reasonable. MotionEye does stress it a lot harder
than most things, but the 53 is a little better. the 60's will cook it over
time.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-20:
@starbasessd the last 2 are my PI NAS which is open
[image]

This is the Motion Eye enclosure 
[image]

What I'm going to have to do is make a larger hole in the top so air can be sucked in

starbasessd, 2020-09-20:
Or mount 1 or 2 fans on the end, there, to suck air through.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-20:
@starbasessd was just thinking that have ordered a new 30mm 5v fan

starbasessd, 2020-09-20:
or 2 20's or 2 15, whatever will best cover the grid there.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-21:
@starbasessd Motioneye crashed last night around 12ish, heres the log
[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/5254912/motioneye.1.log)
also since the rebuild i have noticed that the live playback and also stored video playback plays stuttery

starbasessd, 2020-09-21:
I don't know. I see a bunch of communication and missing drive errors, but
nothing around 00:00. Any disk full or close to full issues? (df -h)
This is a job for @ccrisan! :)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-21:
@starbasessd nope the 60GB usb drive i have hooked up is only at 20% full.

How do i get some assistence from @ccrisan ?

starbasessd, 2020-09-21:
How about the SDCard? how full is it?
I would open a ticket if the old one isn't still available, and post
everything we've done here (images used, settings changed, temps and
throttling data, logs, etc)
I am missing something, but for the life of me, I can't see it...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-21:
@starbasessd how do i check the SD card?
When you say open a tickt do you mean issue? as i havent closed this one yet.

starbasessd, 2020-09-21:
If you have keyboard and monitor access or ssh access type at the prompt
(after login as 'admin' and if prompted for password, it's the same as your
WebGUI password)
df -h <enter>
post results here.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-21:
@starbasessd sorry for the late reply, trying to get another issue im having sorted. Here is the information

Filesystem                Size      Used Available Use% Mounted on
/dev/root               213.0M    188.8M      9.2M  95% /
devtmpfs                911.2M         0    911.2M   0% /dev
tmpfs                   944.2M     16.0K    944.2M   0% /dev/shm
tmpfs                   944.2M      2.3M    941.9M   0% /tmp
/dev/mmcblk0p1           29.9M     12.1M     17.8M  40% /boot
/dev/mmcblk0p3           28.2G    381.0M     26.4G   1% /data
/dev/mmcblk0p3           28.2G    381.0M     26.4G   1% /home/ftp/sdcard
/dev/mmcblk0p3           28.2G    381.0M     26.4G   1% /home/ftp/storage
overlay                  28.2G    381.0M     26.4G   1% /usr
overlay                  28.2G    381.0M     26.4G   1% /var/log
overlay                  28.2G    381.0M     26.4G   1% /var/lib
/dev/sda1                54.8G     22.7G     29.3G  44% /data/media/sda1
/dev/sda1                54.8G     22.7G     29.3G  44% /home/ftp/storage/sda1

starbasessd, 2020-09-21:
Your SDCard is full in the /data folder (where images and movies are
created/stored prior to moving them to your final storage (sda1)

Check the folders /home/ftp/storage for .jpg & movie files you don't need
anymore (probably all of them if you also save to the USB drive.
Command to remove the files is rm -rf (filename or foldername)
If you want the commands, let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-21:
@starbasessd yes please can i have the commands i dont want to mess this up as i am linux command novice

starbasessd, 2020-09-21:
OK
cd /home/ftp/storage <enter>
ls -la <enter>
let me know what that says.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-21:
total 9
drwxr-xr-x    3 root     root          4096 Jan  1  1970 ./
drwxr-xr-x    4 ftp      ftp           1024 Feb  3  2020 ../
drwxr-xr-x    6 root     root          4096 Sep 19 18:01 sda1/

starbasessd, 2020-09-21:
cd /data <enter>
du -ht 1M <enter>
Which path shows the most 1 megabyte or larger files?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-21:
[root@CCTV data]# du -ht 1M
du: invalid option -- 't'
BusyBox v1.29.3 (2020-02-03 23:14:20 UTC) multi-call binary.

Usage: du [-aHLdclsxhmk] [FILE]...

Summarize disk space used for each FILE and/or directory

        -a      Show file sizes too
        -L      Follow all symlinks
        -H      Follow symlinks on command line
        -d N    Limit output to directories (and files with -a) of depth < N
        -c      Show grand total
        -l      Count sizes many times if hard linked
        -s      Display only a total for each argument
        -x      Skip directories on different filesystems
        -h      Sizes in human readable format (e.g., 1K 243M 2G)
        -m      Sizes in megabytes
        -k      Sizes in kilobytes (default)

STUKguy, 2020-09-23:
@starbasessd So its been 2days and the Motion has not crashed, but for some reason i am getting a 10 second delay on the live play back from the cameras this also this include a stutter like effect like its processing frame by frame rather that a smooth feedand the RTSP feed is filled with motion blur and lag. Not sure what to do? do you think i should do another SD card and see if that fixs the problem?

starbasessd, 2020-09-23:
Try lowering resolution, and/or frame rate.(Camera)
Try setting Movie Pass Through to ON (Movies)
Try Turning off Stream Resizing (Video Streaming)
Try adjusting the amount of frames for detection, frames of detection, and
frames after detection.
Try turning off Still Images
BTW, in effect, the system IS processing frame by frame, comparing each
frame to the one before and the one after. This is a SBC, ARM processor,
with at best quad cores at <2.2GHz, and not an Intel or AMD 8-16 cores and
however many threads...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-23:
@starbasessd so for all 3 feeds (2x USB & 1x RTSP) i have done this
Resolution 640 x 480 20FPS
Movie pass through 100% Quality
Video Streaming is off (i dont use it)
Still images are off.
Where do i adjust the frame detection?

starbasessd, 2020-09-23:
Under Motion Detection

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-24:
@starbasessd would a 2gb SD be too small for motion eye? only asking as the system crashed again last night so im going to try another SD card (im hoping the new SD card is not faulty)

starbasessd, 2020-09-24:
Yes, I don't recommend anything smaller than 4GB. The issue isn't
installing the system itself, but all the logs and pics/movie processing
will fill  it almost instantly.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-30:
@starbasessd OK so I'm at a loss went out and brought new 16GB SD card redid Motion OS and still a few days later crashed, plugged back in the previous 32GB SSH'd into it appears that the SDA partition had filled it self up deleted all files and the Motion worked fine for a few days and then crashed SSH'd in again and the SDA partition was filled again. For some reason the SDA partition files are not being deleted after being moved to another partition. Not sure what to do?

starbasessd, 2020-09-30:
Interesting.
What is the path you are using in File Storage (all settings)
What settings are you using in Still Images and Movies?
Are you removing the SDA card without shutting down before removal or
inserting after powering on?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-09-30:
For all the file storage it is set to this
[image]

but when i run df commandi get this
[image]

movies setting is the following, images is turned off (i may need to turn it back on)
[image]

Nope computer 101 make sure power is off fully before removing anything.

side question, how good is your linux commanding/scripting skills?

starbasessd, 2020-09-30:
File Storage
Set custom path
set path to /data/media/sda1/<whatever for each camera>
Try lowering keep time to a lower timeframe (for testing)
If you plug the USB stick or drive into another computer, how big are the
movie files, on average, and how many of them per day? A 64GB drive can
handle quite a few, but if they are very large, and the drive fills very
rapidly, you may want to consider a different storage method, location, or
solution.

As to skills, for personal use, fair to midlin', I can usually get
something done, but I am not a programmer, I am a help desk, who has
delusions of grandeur, why?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-02:
@starbasessd Just noticed the path is set to what you suggested, i have lowered the keep time to 1 day deleted all the files that are already on SDA1. When i plugged my drive into another computer i found some of the files coming from my RTSP feed come in at 2GB-4GB so i removed this feed, see what happens over the next few days, only other option i have in regards to storage would be setting up and FTP which is what i had be before but switched to this set up as i thought it would be more efficiant.

Basically i have set up a power mans UPS for my PI 4 and was hoping to write a scrpit for my PI4 to ping and address or if it loss connection on eth0 for it to safly shutdown, i did find a scrpit online a while back but made the roukie mistake of no saving it :(

starbasessd, 2020-10-02:
OK, while waiting on your results, I'll see if I can find something to do
what you want with regards to connection to network resources.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-02:
@starbasessd I will report back Wednesday just to give it a few days.

I found this https://weworkweplay.com/play/rebooting-the-raspberry-pi-when-it-loses-wireless-connection-wifi/ but obviously for Wifi so im wondering if i can adapt this to eth0?

starbasessd, 2020-10-02:
Something similar built into motionEyeOS.
Settings
Expert Settings
Network Link Watch = ON watches to see if you lose general network
connection and reboots . Doesn't care about ethernet or wifi, will check on
both/either.
Connectivity Watch = ON looks for a specific device on the network or
Internet  (and a specific port) with number of tries before 'failure', time
between tries, etc.
Doesn't care about ethernet or wifi, will check on both/either.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-02:
@starbasessd ah i should have mentioned the poor mans UPS is for my PI4 which is media/file server which is running Raspbian OS this is why im looking at a script to safly shut it down upon loss of network. i will do the same with my motion eye in the future.

starbasessd, 2020-10-02:
Details, details, details....  <grin>
I'll look harder at the solution you posted in a bit.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-02:
Sorry it took me so long to get back to you.
If you have a single connection, wifi or ethernet, that script will work,
and that would be a very easy setup.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-02:
@starbasessd that's OK, I will try the ethernet script and see how it goes, looks easy to me as well just wanted input from someone who knows debian.

STUKguy, 2020-10-05:
@starbasessd So yeah the motion eye crashed (logs attached)

[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/5329567/motioneye.1.log)

The only idea i have left is to dismantle the X823 board i am using and just have the PI, SD,cameras and fan and no SSD for video stroage.

Unless its the temp causing it to crash?

starbasessd, 2020-10-05:
How about the motion.log and the messages log?
When was the last recorded video?
Was there a crash  around
2020-09-28 14:51:31: [motioneye]     INFO: server started
2020-09-30 07:49:15: [motioneye]    ERROR: mjpg client timed out receiving
data for camera 2 on port 8082
or
2020-10-04 13:28:23: [motioneye]  WARNING: 403 GET /login/?_=1601814486946
(192.168.68.110) 2.54ms
2020-10-05 17:00:31: [motioneye]    ERROR: mjpg client timed out receiving
data for camera 2 on port 8082
or
2020-10-05 17:06:59: [motioneye]     INFO: server started
2020-10-05 20:07:47: [motioneye]  WARNING: 403 GET /login/?_=1601924849699
(192.168.68.110) 1.21ms
Interesting that it looks for the camera at 192.168.68.110 and loses
connection.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

[70 more replies]

closed 2020-10-23 by STUKguy