motioneye-project/motioneyeos #2525: Two Pi both delete both their GDrive location subfolders

DrRob, 2020-09-21:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
YES

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
YES

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry PI 2B

### Camera

I am using the following type of camera: MMAL

My camera model is: Pi Noir v2 on one and Pi v1 on the other

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi on one and Ethernet on the other

### Peripherals

One has a USB Wi-Fi adapter, but I don't believe that's relevant

### Log Files

motioneye.log from the last reboot:

2020-09-21 07:43:27: [motioneye]     INFO: hello! this is motionEye server 0.42.1
2020-09-21 07:43:30: [motioneye]     INFO: cleanup started
2020-09-21 07:43:30: [motioneye]     INFO: wsswitch started
2020-09-21 07:43:30: [motioneye]     INFO: tasks started
2020-09-21 07:43:30: [motioneye]     INFO: mjpg client garbage collector started
2020-09-21 07:43:30: [motioneye]     INFO: smb mounts started
2020-09-21 07:43:30: [motioneye]     INFO: server started
2020-09-21 07:45:54: [motioneye]     INFO: gdrive: found 29/2 folder(s) in local/cloud
2020-09-21 07:45:56: [motioneye]     INFO: gdrive: deleted a cloud folder 'front'
2020-09-21 07:45:57: [motioneye]     INFO: gdrive: deleted a cloud folder 'rear'
2020-09-21 07:45:57: [motioneye]     INFO: gdrive: deleted 2 cloud folder(s)
 
Both of my Pi are uploading to GDrive, one to location "cctv/front" and one to location "cctv/rear".  Both are set to "Include Subfolders" and "Clean Cloud".  One is set to keep images and movies for one month, and the other is set to keep movies for one month and images for one day.

This setup worked well for a while (maybe a month, I'm not sure), but at some point the "front" and "rear" folders (but not the "cctv" folder) started being deleted (along with their entire contents, obviously).  The motioneye.log on both Pi indicate that both of them are doing this.

Why are they deleting the location folder, and why is each of them deleting the other's location subfolder?

I will try moving the "front" and "rear" locations up to the root of the GDrive to see if that works around the problem.

starbasessd, 2020-09-21:
Are they linked in any way? To each other, (each shows both in the GUI) or
to a hub that shows both, that were connected as motionEye Cams?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DrRob, 2020-09-21:
They're only linked in that they're connected to the same local network and uploading to the same GDrive (but different locations).  They haven't been configured to know about each other, and there isn't another MotionEyeOS to group them either.

I've altered the configured GDrive locations so that the first part of the path isn't in common (i.e. from "/cctv/front/" and "/cctv/rear/" to "/front/" and "/rear/"), and so far after a few hours the folders haven't been deleted.

starbasessd, 2020-10-01:
If it is resolved, please close.

DrRob, 2020-10-02:
The problem can be avoided by not having two motionEyeOS uploading to GDrive with a common prefix in the path.  I don't think I'd count that as "resolved", but it's good enough for me so I'll close the ticket, thanks.  (Maybe mention it in the docs?)

closed 2020-10-02 by DrRob

starbasessd, 2020-10-02:
Will do, thanks.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-05:
To clarify, you were using two separate motioneyeos boxes pointing to the
same gdrive structure, correct?
motioneyeos1 with camera1 pointing to gdrive/motioneye/camera1
motioneyeos2 with camera2 pointing to gdrive/motioneye/camera2
correct?
And your change was to point them
motioneyeos1 with camera1 pointing to gdrive/camera1
motioneyeos2 with camera2 pointing to gdrive/camera2
correct?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DrRob, 2020-10-05:
Yes, that's correct.