motioneye-project/motioneyeos #2526: Can't Access dev20200907 After Installation

Dewey3, 2020-09-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev2020907

### Board Model

I am using the following board/model: Raspberry Pi 3B+

### Camera

I am using the following type of camera: V4L2 Camera

My camera model is: ELP 1 megapixel USB Camera

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: N/A

### Log Files

I consider the following log files relevant to this issue: N/A

###

I decided to give dev20209007 a try on one of my cameras.  I blew the image out using Windoze 10/Win32DiskImager as I always have, and before removing the freshly imaged card, wrote my usual wpa_supplicant.conf and static.conf files to the card so I could do an in-place headless setup:

**wpa_supplicant.conf:**
````
country=US
    update_config=1
    ctrl_interface=/var/run/wpa_supplicant

    network={
        scan_ssid=1
        ssid="mySSID"
        psk="myPASSWORD"
    }
````

**static.conf:**
````
STATIC_IP="192.168.1.209/24"
STATIC_GN="192.168.1.1"
STATIC_DNS="8.8.8.8"
````

The problem is that I can see the camera attached to the network and even successfully ping the camera, but I cannot access it.  I've tried by typing the IP address into a Chrome browser, which does not see the camera at all.  I've also tried ssh'ing directly into the camera with Windoze/PuTTY and from the Ubuntu 20.04 server, both which result in "connection refused".  I'm sure it's probably something I'm either missing, or doing wrong from my usual 20190911 cameras, but am stumped for now.  Any ideas???

starbasessd, 2020-09-22:
FIlename is 'static_ip.conf'
Do you have the file 'ssh' or 'ssh.txt' also in /boot? when you imaged the
SDCard?
Do you have access to console? (keyboard and mouse attached)
if so, what does:
ifconfig <enter>
show for inet line under wlan0?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Dewey3, 2020-09-22:
> 
> FIlename is 'static_ip.conf' Do you have the file 'ssh' or 'ssh.txt' also in /boot? when you imaged the SDCard? Do you have access to console? (keyboard and mouse attached) if so, what does: ifconfig <enter> show for inet line under wlan0?
> […](#)

My mistake on leaving the "_ip" out of my original post, the file static_ip.conf is properly titled and working as evidenced by being able to see the camera attached and being able to ping it at the '209 ip address.  In the past, I've only written the wpa_supplicant.conf and static_ip.conf files to the card for the headless setup, and never written a ssh or ssh.txt file before.  What file should I be using, ssh or ssh.txt, and what should go in the file?  Thanks much!

starbasessd, 2020-09-22:
BTW, you may be missing
key_mgmt=WPA-PSK
from your wpa_supplicant.conf file
Here's a good link to all the config files:
https://github.com/ccrisan/motioneyeos/wiki/Configuration
on the right, under "Configuration"...
-----
[root@meye-c0e188a9 ~]# cat /data/etc/wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="ssid"
    psk="password"
        key_mgmt=WPA-PSK
}
-----
If you are not in the US, make sure to set to your country code.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-09-22:
> (keyboard and mouse attached) if so, what does: ifconfig show for inet line

either works, and there's nothing in it. It just looks for the file before
turning on the ssh server.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Dewey3, 2020-09-22:
Thanks.  So far, this one really has me scratching my head.  It can't be a network connection issue as I see it attached to the home router, and can successfully ping it.  The problem is that I can't connect to it... not with a browser, not with PuTTY, and not ssh'ing from Ubuntu Server 20.04.  I know the PuTTY and ssh attempts are working, as I am getting "Connection refused" with both of those as opposed to a time out in attempting to connect (I did add a blank ssh.txt file to see if that would work).  I really hate to pull the Pi down to hook it up to a keyboard and screen, but that is what I may have to do so I can see what's going on.  As always, thanks!

starbasessd, 2020-09-22:
Sounds like the ssh server may not have started.
when you get a chance, you can run
netstat -apt | grep sshd <enter>
to confirm SSH Server is running.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-09-22:
Oops, forgot one thing:
You have access to the WebGUI? Go to
Services, and at the bottom, is Enable SSH Server ON?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Dewey3, 2020-09-22:
Yea, that (Chrome browser) is not working at all, as if the camera doesn't exist.  The only thing I'm doing different from all of my 20190911 installations is that I used Win32DiskImager v1.0 to blow out the image, where I used v0.9.5 in the past.  I haven't read any reports of the newer Win32DiskImager introducing bugs, but I may have to reinstall the older version since they can co-exist, just to see if that is somehow the problem.

starbasessd, 2020-09-22:
FWIW, I only use Win32DiskImager for production machines.
Have you tried Firefox or Edge?
My Google Chrome Version:
Google Chrome is up to date
Version 85.0.4183.121 (Official Build) (64-bit)
Working here.
MotionEyeOS is http://<ip_address>
motionEye on another OS is http://<ip_address>:8765
ssh  is admin@<ip_address> if you use Linux Shell in Win 10

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Dewey3, 2020-09-23:
Ok, I pulled the Pi and hooked it up to a monitor so I could see the first boot, and I've found the problem.  When this new version (dev20200907) boots, I see it successfully "Configuring wireless network".  However, after that, I see "Configuring wired network", and since I'm doing a headless, wireless install, it doesn't see a wired network and goes into a panic rebooting loop over and over and over again.  My task now is to find out how to get it to ignore looking for the wired network on first boot.  I've tried a os.conf file with `OS_NETWORKLESS="true"`, but that didn't work.

starbasessd, 2020-09-23:
It's not seeing the wireless connection, either.
Double check your wpa_supplicant.conf file.
SSID and Password are case sensitive, needs quotes, and should look like
this:

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="ssid"
    psk="password"
key_mgmt=WPA-PSK
}

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Dewey3, 2020-09-23:
You've mentioned it not seeing the wireless connection before, but I still respectfully disagree since I can see it attached to my home router, and can ping it.  (I think the pings were successful because the wait time grows each time it panic reboots, which was giving the pings chance to connect).  I did a troubleshooting workaround for the panic rebooting by connecting it via wired connection, and of course it booted up fine.  Once I did that first successful boot, I disconnected the ethernet cable and rebooted and can now connect wirelessly and remote in.  So my first boot/panic rebooting problem can be solved by booting the very first time hooked into a wired connection.

However, now that I've got it working wirelessly, I discovered a new problem that may keep me from upgrading any further.  Once I had the wireless connection working as expected, I placed the Pi back in position and hooked up the camera.  I added the camera, but it is not giving me a picture.  The Pi/MotionEye are seeing the camera, but giving me a gray screen with the camera name and time.  I deleted it and re-installed, rebooting each time, with the same results.  So unless this is a bug that is quashed in a later release ("dev" issues are of course a fluid development), I will be locked into 20190911 since my entire system consists of the ELP 1MP USB cameras ([https://www.amazon.com/gp/product/B00VFLWOC0/](url)).

Dewey3, 2020-09-23:
Ok, my final report for now.  Wanting to troubleshoot the proper way, I started completely from scratch without adding the wpa_supplicant.conf and static_ip.conf files, and executing the very first boot while connected via wired network.  I am happy to report that of course, it booted fine **and** saw my camera.  So I'm guessing that after all of the panic rebooting and without a camera attached when first trying, I could have easily caused the gray screen problem.  So for now, the only real problem that I am seeing with this release is that it will not do a wireless first boot.

starbasessd, 2020-09-23:
Did you make the changes to the wpa_supplicant.conf file? added
country=<your country code>, and key_mgmt=WPA-PSK? THere have been updates
to the kernel and other systems that require a country code to be set but
not in the wiki documentation.
The dev20200907 image fixes issues in the RPi4 USB images that prevented
USB cameras from working if only updating the start4 and fixup4 files for
20200606, 20200203, and 20190911 images (among other things)
I would try deleting the camera, move the USB port, reboot, and re-set up
the camera.
I do not have any issues with dev20200907 and Pi4s or Pi3s with USB camera.
Does your camera have outside power, or does it draw from the PI? If it
pulls from the PI, try clicking shutdown, unplugging the power from the PI
and re-plugging instead of rebooting.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Dewey3, 2020-09-23:
Ok and just to see, I wrote the wpa_supplicant.conf just as you dictated the first time (the indenting on your first and second key_mgmt switches are different).  It doesn't matter, I'm still getting the exact same startup sequence and panic rebooting result as when I was using my 20190911 wpa_supplicant.conf file.  The booting can be seen here: [https://drive.google.com/file/d/1m_-PgMJklnoDp_8clCeLJyFl56o5yUi2/view?usp=sharing](url).

starbasessd, 2020-10-27:
BTW, looking over this issue, I saw in your static_ip.conf file you have:
STATIC_GN="192.168.1.1"
it needs to be 
STATIC_GW="192.168.1.1"