motioneye-project/motioneyeos #2528: How do I see a playback video (i  in Motion EyeOS  ?

aksa59669, 2020-09-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-09-23:
Click on the stream view
It will open a menu bar
Click on the triangle top right
Look for the file time and date and click on the preview

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

aksa59669, 2020-09-24:
@starbasessd 
your answer is not suitable for my question
I have choosen to record 24 hours of video and wanted to view the recording after 1 hour the recording started ,It displaying the following(see the images attached) issue.
solve it please..
[image]
[image]

starbasessd, 2020-09-24:
My answer to your original request was correct.
As to your edited question, you can't, the file is still open by the program for writing.

aksa59669, 2020-09-24:
@starbasessd 
Can you please answer for the 2nd comment
I didn't understand your reply

starbasessd, 2020-09-24:
Simple answer, you can't.
The file is locked by the program writing the file to the drive. It cannot be opened by a second program at the same time.
There are programs that can allow write and read at the same time, but this is not one of them.

aksa59669, 2020-09-24:
@ccrisan
I request the developer to bring this feature

closed 2022-01-19 by starbasessd