motioneye-project/motioneyeos #2529: Videos stop being save on SMB Serveur

AymericFR02, 2020-09-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi 4 B 2Gb.

### Camera

I am using the following type of camera: MMAL.

My camera model is: Raspberry . Pi NoIR Camera Board V2, 8mp

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue: 

no peripherals

### Log Files

I consider the following log files relevant to this issue:

I've search in the log but I didn't find something, but i attached you the log files

[motioneye (4).log](https://github.com/ccrisan/motioneyeos/files/5266127/motioneye.4.log)

----------------------------
 
Hi everybody,

For you information, I'm French so, sorry if I made some mistakes.

I install in august, a raspberry Pi 4 B with the Raspberry Pi NoIR Camera Board V2. 
I configure MotionEyeOS for saving on a Synology NAS by the SMB protocol.

Everythings worked perfectly but one day the videos stop being save on the NAS.

I update MotionEyeOS from 20190911 to 20200606 version.
But it doesn't fixed the issue.

There is free space on the NAS.

When I reboot the Raspberry, one video is saving but not the next one. 

The video saving is configure to save video of 1 hour in "still recording".

The streaming work perfectly.

Do you have an idea to resolve this problem ? 

Thanks.

starbasessd, 2020-09-23:
Please try to install dev20200907 found here:
https://github.com/ccrisan/motioneyeos/releases
or downgrade back to 20190911.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AymericFR02, 2020-09-23:
The problem wasn't appear after the upgrade. 
I upgrade because of the problem. 

So, do i really need to downgrade or upgrade the Raspberry ?

starbasessd, 2020-09-23:
I would try the newer one, then. dev20200907

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AymericFR02, 2020-09-23:
Ok, I'll try.

But is it possible to do it by SSH ? or I need to make a fulll install of the system ?
Because I can't access to shared file of my Raspberry, I don't have the login and the password.

starbasessd, 2020-09-23:
If you have access to the WebGUI, you can turn on Enable Pre-Releases Under
Expert Settings, then do a Check under General

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AymericFR02, 2020-09-23:
Yes, it's done.

I test and I tell you if it's work. 

Thanks

closed 2020-09-23 by AymericFR02