motioneye-project/motioneyeos #2531: Camera won't startup with RPi4

yuniq-neko, 2020-09-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Raspberry Pi 4 (Model B - 2GB RAM)

### Camera

I am using the following type of camera: MMAL

My camera model is: Camera Module V2

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Just the Pi4 + Camera Module powered by official raspberrypi USB-C power supply

### Log Files

I consider the following log files relevant to this issue:

[motion.log](https://github.com/ccrisan/motioneyeos/files/5268705/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5268706/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5268708/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5268709/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5268710/dmesg.log)

yuniq-neko, 2020-09-23:
I have both a Pi3 and a Pi4 running same version of MotioneyeOS; Pi3 works perfectly fine but for some reason, the Pi4 does not like the camera at all; MotionEyeOS shows the camera but says its "offline" nomatter what i do :/... looks like motion is panicking in the background too a bit...

yuniq-neko, 2020-09-23:
By offline; I mean that it shows this thumbnail instead of Camera image:
[image]

yuniq-neko, 2020-09-23:
Its also seeming to try to reboot itself but also dumping stack trace of some sort whilst doing so...
This is all I can catch of the dump:
[image]

starbasessd, 2020-09-23:
dev20200907 works for Pi4, previous versions need a manual update to
start4.elf and fixup4.dat (but will have issues with USB cameras).

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-09-23:
Here's the 2 files.
[RPi4-Boot-Files.zip](https://github.com/ccrisan/motioneyeos/files/5268765/RPi4-Boot-Files.zip)

yuniq-neko, 2020-09-24:
I updated these 2 files and it has fixed the spontaneous crash stack and reboot... however camera still persists to not want to work 🤨 It does show "video" output now, but it is just motion displaying a grey screen with "Camera Unreachable since ..." in the top right :/

Updated logs attached: 
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5275507/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5275508/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5275510/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5275511/dmesg.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5275512/motion.log)

starbasessd, 2020-09-24:
Is the camera the new HD one? If so, please try the dev20200907 image. The
new HD camera wasn't supported by the kernel in the older images, but was
one of the updates for the dev20200907 image.
If it is the older one (no detachable lens) then try lowering the
resolution to 640x480 and low framerate, and work your way up to higher
resolutions and frame rates.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

yuniq-neko, 2020-09-25:
it is not the new one, Tis the v2 with no removable lens :/...
dropped the resolution+framerate as far as it went and still no luck. :/
Camera LED doesnt even light up...
I'm starting to think this is a bigger issue related to Pi4 as I tried the camera on normal Raspbian and it also wouldn't work with anything there... only when I attached it back to a Pi3 is when it started working with everything again. :/

starbasessd, 2020-09-25:
Interesting.
Which version of Raspbian did you try (please don't say 'the latest', give
the date of it) and what application(s) did you try?
In Raspbian, did you turn the camera 'on'?
In raspi-config you need to enable the camera, and i2c and spi to enable
the camera.
 (They are enabled in motionEyeOS)
In Raspbian, after enabling try raspivid & raspistill programs
https://projects.raspberrypi.org/en/projects/getting-started-with-picamera/3
Let me know...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

yuniq-neko, 2020-09-26:
solved the problem after noticing the issue suddenly happen on the Pi3 - Twas a dumb one... Theres a split thats begun to form on the ribbon flex that connects the camera to the Pi, after testing with a multimeter, theres poor continuity on some of the lines. 
I have just replaced the ribbon flex and just like magic, the camera now works on the Pi4. Sorry to have wasted your time with this issue @starbasessd.

closed 2020-09-26 by yuniq-neko