motioneye-project/motioneyeos #2532: How to nano /boot/config.txt

TheTruthIsPie, 2020-09-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Pi4 4GB

### Camera

I am using the following type of camera: Pi v2 camera

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

SSH

### Log Files
This might be a trivial question to all the Linux super users, I would like to modify the motionOS config.txt file, but when I ssh to the camera, and log in as admin (or even as root) and do sudo nano /boot/config.txt 
I get this 
`-sh: sudo: command not found`
I tried su
`su: unknown user nano`
I do apologise I am not a power user, but how can I modify the config.txt

When I try without sudo I am not able to save the file
I am trying to disable all the leds That worked on my RaspbianOS using this in the config file:
# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
# Disable the PWR LED.
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off
# Turn OFF Network LED's
dtparam=eth_led0=14
dtparam=eth_led1=14

Tried searching on the issues here but noone seems to have this issue which leads me to believe I have done something wrong.

Thanks

starbasessd, 2020-09-23:
When you are ssh'd in (or even on the console, you ARE root/admin and don't
need "sudo" for anything... You need to turn debugging ON in the WebGUI
(Under Expert Settings)(easiest)(will prompt to reboot)
When completed, you can turn Debug back off.
If all else fails, though you can edit the file right after installing an
image, the /boot partition is where the file resides. When I image, it's my
E: drive (Win10)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheTruthIsPie, 2020-09-26:
Thank you so much, such a simple fix, the debugging make it work straight away.

starbasessd, 2020-09-26:
Be sure you turn it back off to protect the files from accidental
corruption.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DaniW42, 2020-10-01:
I am not pretty sure, but this could also happen because you (or motioneyeos) is using Shell(sh) instead of the more common Bash

DaniW42, 2020-10-01:
Well, this might be true and would most likely (I am not using motioneyeos) be the next question one could ask. But it is, somehow, not related to the problem described in the issue... Anyway, looks like it was fixed and it is working. 

This should be closed.

closed 2020-10-01 by TheTruthIsPie