motioneye-project/motioneyeos #2533: No Network Share under File Storage when using a Docker

teddy9264, 2020-09-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 0.42

I am running MotionEyeOS in a docker on my Ubuntu Server 20.04. I don't know much about dockers and I just followed the guide on the wiki.

My issue is that there is no option to have file storage as a network share, only custom path like there are on different versions of the OS on Raspberry Pi.

At the moment I believe the only way around it would be to mount directories between the host and docker but I wasn't able to do that.

I'm unsure if this is a bug report or a feature request.
[image]

starbasessd, 2020-10-01:
What is your docker host setup?
Machine, OS, resources (RAM/Drive/network connectivity), etc. 
I have time to build/play with a setup to help you out.

teddy9264, 2020-10-01:
I have an i5 7600, Windows 10 Enterprise, 16gb DDR4 RAM, My OS is off an NVME M.2 960 Evo 250gb SSD but the VM is on my 2tb hard drive. My computer uses WiFi, with my VM in bridged mode, and is set up with a static IP that works for everything else I use my server for.

I'm not sure if this is an issue with my computer or network share isn't includes on dockers as the 'Network Share' is included on the Raspberry Pi images. But any help is greatly appreciated!!

starbasessd, 2020-10-02:
What kind of network share? SMB, FTP, NFS, something else?
If SMB, what level?
Trying to connect via name or ip address?
name or name.local?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

teddy9264, 2020-10-02:
I have an SMB share, I believe SMB3 (but I'm not sure). I connect through an ip. My server is on the same machine as the docker so I have been testing by connecting to a windows share on a separate computer to see if that's an issue- hasn't fixed anything so I think I can't connect to a network location through 'custom path'.

starbasessd, 2020-10-02:
If you use custom path did you use
\\<ipaddress>\[sharename] for the path?
or mount the share locally and then access with the path
\[root]\
Sorry for all the questions. I'm not a fan of Docker, but willing to try to
learn to support another system.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

teddy9264, 2020-10-02:
I did \\<ipaddress>\[sharename]. I don't know how to mount the share but that might be the way to go.

starbasessd, 2020-10-02:
> I did \[sharename]. I don't know how to mount the share but that might be

Open File Explorer
for location (address bar) type:
\\ipaddress\ <enter>
Should open shares
right click the shared name
Click Map Network Drive
Enter appropriate information (drive letter, folder, etc...)
Click Finish
In custom path select for correct option, or enter drive_letter:\path (
m:\motioneye)
Let me know if that works for you. Still building a VM of Win10...)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

teddy9264, 2020-10-02:
Sorry, my host is a Windows 10 computer with my guest being an **Ubuntu Server**, so I can't map on windows for it to work. So I have the MotionEyeOS docker running on my Ubuntu Server which is a VM on my Windows 10 PC.
I'm unable to test now but I will try and mount the network location in the docker but I'll have to google how to do that.

starbasessd, 2020-10-02:
You can, but you need to enable Shared Folders in the Ubuntu VM.
Using VirtualBox or VMWare/eSXI/Player?
Which Ubuntu Server? 16.x, 18.x, 20.04? Desktop or Server?
Was 80% into building a Win10 VM on my Win10 Host. Got a few Ubuntu Server
images I can clone, though...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

teddy9264, 2020-10-02:
VMWare Player Ubuntu Server 20.04 Server, I can do that but aren't dockers sperate from my Ubuntu Server?

starbasessd, 2020-10-02:
Yes, and no. (Clear as Mud? :) )
It becomes a VM on your VM. If you Connect a Shared Folder (not sure how in
VMPlayer) you would then access in Ubuntu like a local folder....
Let me get a VM fired up...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

teddy9264, 2020-10-02:
I can set up shared files in VMWare, but I don't know how to access it in Ubuntu let alone the docker!

starbasessd, 2020-10-02:
Assuming it's similar to VBox, it mounts the share in
/media/(user)/shared_name)  [/media/kevin/shared_sf/]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-02:
OK.
Very interesting.
Win10 host. Ubuntu 20.04 Guest in VBox. Shared folder to Host network
mapped drive. Validated shows up in Ubuntu as /media/sf_sdc/.

docker pull ccrisan/motioneye:master-amd64

then ran
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-amd64

Was able to connect to docker instance from Ubuntu desktop using FF
and http://localhost:8765

Added camera, able to view it

I put (what I thought was my path to a test folder) in Custom Path.
/media/sf_sdc/motioneye/
Turned on manual snapshots under Still Images
took a couple snapshots
Confirmed they were in there in the built in viewer
went back to command line, looked for them in /media/...
Nada.
went to command line, changed to /var
ran sudo find -name *.jpg
found them in
/var/lib/docker/overlay2/<long docker identity
string>/diff/media/sf_sdc/motioneye/<date&time.jpg>
Not what I was expecting.
Will look further on how to expose the underlying file system to the docker
instance, or if at all possible.
Interesting.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-02:
OK
found instructions here:
https://www.digitalocean.com/community/tutorials/how-to-share-data-between-the-docker-container-and-the-host
so, the way I read it, I should be able to add a line to the startup like
-v /path:/path/in/container
so if I add
-v /media/sf_sdc/motioneye:/media/motioneye
and then add the Custom Path
/media/motioneye
It should work
Let you know in a few...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-02:
OK
Geez, they don't make this easy...
My startup for the docker container is set to:
docker run --name="motioneye"  \
  -p 8765:8765  \
  --hostname="motioneye"  \
  -v /etc/localtime:/etc/localtime:ro \
  -v /etc/motioneye:/etc/motioneye  \
  -v /var/lib/motioneye:/var/lib/motioneye \
  -v /media/sf_sdc/test:/media/test  \
  --restart="always"  \
  --detach=true  \
  ccrisan/motioneye:master-amd64

the line
  -v /media/sf_sdc/test:/media/test  \

creates the link of the folder in the Ubuntu machine, to the folder in the
docker session.
which now works as expected. (The Ubuntu session maps the Win10 network
mapped drive (mapped to a network share) to /media/sf_sdc)
Holy *&(%($((.
BTW if you get errors about a /motioneye folder already in use, I fixed by
using command
docker rm motioneye <enter>
then rerunning the new start command.
I might clean this up, and make notes for the documentation tomorrow, now I
have a headache and off to bed.
Donations to my local psych facility might be nice, too...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

teddy9264, 2020-10-02:
Thanks for this I will try it, although this is quite a work around where if the developers added a network share setting to the docker would be a lot easier! Thanks again.

starbasessd, 2020-10-02:
There may be, I'm just beginning to learn docker. Might ask in a docker
forum/support group. If yo udo, let me know here, too, please so I can
document it...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-02:
Having played with it for a while, I'm not seeing an easier option than the
one I worked out.
In order to add cifs(SMB), you would have to do the build option and add it
there.
The way I showed (adding another -v option (mapping a folder from the
Docker Host) is a lot easier,
And mapping a Shared Folder from the Win VM Host to the Ubuntu Guest (which
is the Docker Host), at least in VirtualBox, was still easier.
If I get time this weekend, I'll see how far I can get, and I will see if
@ccrisan would be interested in adding cifs support to the docker port.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2020-10-02:
> I will see if @ccrisan would be interested in adding cifs support to the docker port.

I don't really have time to take care of this task at this moment. PRs are welcomed.

starbasessd, 2020-10-02:
No problem, thanks for letting me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-02:
Building a copy doesn't look like it's going to happen.
Followed part 2 of
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
and received this error
[image: image.png]
Still researching.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

teddy9264, 2020-10-02:
Yeah, I had a try with building my own docker, might try that again and see if I can get different results.

starbasessd, 2020-10-23:
Status?

NdR91, 2020-12-03:
I have the same issue. In my case, MotionEye is running on docker on a laptop with Ubuntu 10.04.
If i modify the "motioneye.conf", from  "smb_shares: false" to "smb_shares: true", ME doesn't start anymore with the following log:

```CRITICAL: please install cifs-utils```

Which is already installed in my machine.

Also, on the "smb_shares" settings it says I have to run the container as root, which is not very clear to me (I started few weeks ago with docker). I'm also using portainer to help me with a GUI.

Do you have some suggestions?

starbasessd, 2020-12-03:
The best work-around I've found is to mount the shares wanted on the docker
Host, then share to the docker via a -v /media/share:/media/share
connection (or whatever path you want).
I, too, am new to dockers, and have on my to-do list build my own version
and document it. CCrisan has too much on his plate, and asked for PRs
assistance...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?